stuartpittaway/diyBMSv4Code #3: Missing sdkconfig.h, freertos/semphr.h, esp_task_wdt.h, ...

Klauzen, 2020-02-15:
Hi!
I try to get this debug code up running, but after cloning the git, the compiler ask for some missing libs: sdkconfig.h, freertos/semphr.h, esp_task_wdt.h. I tried to install the libs manually, without success.

`In file included from .pio\libdeps\nodemcuv2\AsyncTCP_ID1826\src\AsyncTCP.cpp:24:0:
.pio\libdeps\nodemcuv2\AsyncTCP_ID1826\src\AsyncTCP.h:26:23: fatal error: sdkconfig.h: No such file or directory`

How can I install the right libs?
Thanks!

stuartpittaway, 2020-02-15:
Hello, that's an upstream library dependency.

Platformii should automatically install the correct version, perhaps try a clean before building?

Klauzen, 2020-02-15:
Hi Stuart,
thanks for replying. But I already tried to clean the project, without success... The error still appears!

stuartpittaway, 2020-02-15:
Just checking you are using the code from the new repo?

https://github.com/stuartpittaway/diyBMSv4Code

The automatic Travis build is working fine, so unfortunately must be an issue with the installation of platformio on your pc

Klauzen, 2020-02-15:
Hi, yes, this is the code form the repro. But there is no chance to get this code working. This is strange, because the normal ESPController code is building perfectly...

Klauzen, 2020-02-15:
Ah, found a solution: It's a known bug in PlatformIO since yesterday (after updating).

Just add
`lib_ignore = AsynTCP
lib_compat_mode = strict`
in the platformio.ini and it will work.

closed 2020-02-15 by Klauzen