stuartpittaway/diyBMSv4Code #10: ESP8266 with 4 MB flash memory - WIFI Password issues

stuartpittaway, 2020-03-16:
If you have an ESP8266 with only 4MB flash memory the controller will enter a loop when configuring the wifi setup - it appears like the password isn't stored (its not!) so the controller attempts to go into a setup loop.

The recommended ESP8266 is the D1 Mini Pro - this has 16MB Flash by default.

buy from here https://amzn.to/2OF7Mzn

To make DIYBMS work on 4MB flash - change the option in platformio.ini 

> build_flags = -Wl,-Tesp8266.flash.16m15m.ld

to

> build_flags = -Wl,-Tesp8266.flash.4m1m.ld

Note that future versions of the controller code may require larger flash sizes to store other information, historic logging etc.

davidbue, 2020-06-13:
Greetings from Denmark,

I have just started my journey into the diyBMS V4 and while I am waiting for my pcbs to arrive I thought that I might as well take a shot at setting up the code on a Wemos D1 mini 4MB that I had laying about.

Initially I was stuck in the boot loop but the above fixed the issue for me. Now a new issue has come up. I can setup the SSID, and upon rebooting the Wemos seem to disappear completely. It does not join my home wifi, nor does it broadcast its own.

When trying to debug I cannot get the serial communication up running, so for good measure I completely erased the flas of the wemos and uploaded the code again. Now it boots but broadcasting its own SSID, yet still I get no output at all on the serial port.

Cheers,
  David

stuartpittaway, 2020-06-13:
See issue 10 in the other GitHub repo.

davidbue, 2020-06-26:
Dear Stuart,

I am not certain that I understand what repo to go to for understanding this issue. I just went to the hardware diybms hardware github, but there are only 4 posts there.

Could you please point me in the right direction? :-)

Thanks!

stuartpittaway, 2020-06-27:
Do you not yet have the controller PCB?

davidbue, 2020-06-28:
I have all the hardware with the SMD's mounted from JLCPCB, and a 3.3v FTDI USB serial dongle. I managed to get it all up running tonight. 

The issue was so simple:

Wrong wifi passphrase typed in at first boot made the Wemos subsequently boot up and fail to connect via wifi after which it got stuck.

The reason why that I got no data from Serial1 was (facepalm) a bad test lead.

Alas, everything on the Wemos works now. Only thing that I am missing is the ATTINY for the balance boards. It's arriving Tuesday or Wednesday after which I will be able to hopefully test the completed system. 

Stuart, thank you so much for making this BMS available. I appreciate it very much.

[image]