stuartpittaway/diyBMSv4Code #11: The controller is having difficulty communicating with the cell monitoring modules.

gen-ben, 2020-04-05:
Hello, I have a problem with communication, the esp8266 "4MB" can not connect to cell models. All cell modules seems to work fine in standby mode the blue LED double flash every 8 second and bypass work when cells voltage exceed 4.1V. But when I connect the TX from ESP to RX in cell model all LEDs turn off in the Module and nothing happen after. I am getting this error from ESP: "The controller is having difficulty communicating with the cell monitoring modules." I have double checked the connection everything seems fine and I tried to connect only one model it did not work too. Any ideas to make it work??

stuartpittaway, 2020-04-05:
What controller PCB are you using?

gen-ben, 2020-04-05:
I am using my own PCB layout I have build it using the schematic in: https://github.com/stuartpittaway/diyBMSv4

gen-ben, 2020-04-05:
I have double checked the PCB it is identical to the design in the link above.

stuartpittaway, 2020-04-05:
What optoisolators are you using? Have you check there are no shorts?

As you have built your own design I can't really help as much as I could normally

gen-ben, 2020-04-05:
I am using "HMHA2801" optocouplers , yes I did check for shorts non found.
I have attached PDF file contain few SS for used PCB layout. Please check it, is there any thing wrong with this layout?.
[1.pdf](https://github.com/stuartpittaway/diyBMSv4Code/files/4434335/1.pdf)

stuartpittaway, 2020-04-05:
So you have rewritten the entire PCB board for both controller and modules? I don't have the time to check the design, sorry. Why did you need to redesign it?

gen-ben, 2020-04-06:
I am new to this stuff, I wanted to learn a PCB design software so I redraw everything, and I wanted to get rid of the cabling between the PCBs by using switches instead.
do you think such problem is from the control or cell monitoring circuit?.

gen-ben, 2020-04-06:
Here is the log from serial1 with one module connected:

Apply default config
Found pcf8574
Connecting to WIFI
Connecting to Wi-Fi...
mode : sta(50:02:91:46:a6:45)
                             add if0
                                    scandone
                                            state: 0 -> 2 (b0)
                                                              state: 2 -> 3 (0)
                                                                               state: 3 -> 5 (10)
                 add 0
                      aid 1
                           cnt

                               connected with XperiaXZ1, channel 11
                                                                   dhcp client start...
       ip:192.168.43.87,mask:255.255.255.0,gw:192.168.43.244
                                                            Connected to Wi-Fi.
3. Connected IP:192.168.43.87
Requesting NTP from time.google.com
Got NTP time: 03:53:47 06/04/2020
S:80/1/1=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925/Q:2
Rules:0000000100=
S:80/3/2=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7301/Q:1
S:80/4/3=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CA8E/Q:0
S:80/1/4=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C5D6/Q:1
S:80/3/5=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9799/Q:0
Rules:0000000100=
S:80/1/6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EDBD/Q:1
S:80/3/7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BFF2/Q:0
pm open,type:2 0
                Rules:0000000100=
S:80/1/8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =34AC/Q:1
S:80/3/9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =66E3/Q:0
Rules:0000000100=
S:80/1/A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1CC7/Q:1
S:80/3/B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4E88/Q:0
S:80/1/C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =647A/Q:1
Rules:0000000100=
S:80/3/D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3635/Q:0
S:80/1/E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4C11/Q:1
S:80/3/F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1E5E/Q:0
Rules:0000000100=
S:80/1/10=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C679/Q:1
S:80/3/11=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9436/Q:0
Rules:0000000100=
S:80/1/12=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EE12/Q:1
S:80/3/13=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BC5D/Q:0
Rules:0000000100=
S:80/1/14=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =96AF/Q:1
S:80/3/15=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C4E0/Q:0
S:80/1/16=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BEC4/Q:1
Rules:0000000100=
S:80/3/17=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EC8B/Q:0
S:80/1/18=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =67D5/Q:1
S:80/3/19=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =359A/Q:0
Rules:0000000100=
S:80/1/1A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4FBE/Q:1
S:80/3/1B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1DF1/Q:0
Rules:0000000100=
S:80/1/1C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3703/Q:1
S:80/3/1D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =654C/Q:0
Rules:0000000100=
S:80/1/1E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1F68/Q:1
S:80/3/1F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4D27/Q:0
S:80/1/20=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =33F2/Q:1
Rules:0000000100=
S:80/3/21=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =61BD/Q:0
[... 74 lines cut ...]
Rules:0000000100=
S:80/1/1C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3703/Q:1
S:80/3/1D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =654C/Q:0
Rules:0000000100=
S:80/1/1E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1F68/Q:1
S:80/3/1F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4D27/Q:0
S:80/1/20=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =33F2/Q:1
S:80/3/21=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =61BD/Q:0
Rules:0000000100=
S:80/1/22=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1B99/Q:1
S:80/3/23=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =49D6/Q:0
Rules:0000000100=
S:80/1/24=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =6324/Q:1
S:80/3/25=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =316B/Q:0
Rules:0000000100=
S:80/1/26=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4B4F/Q:1
S:80/3/27=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1900/Q:0
Rules:0000000100=
S:80/1/28=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925E/Q:1
S:80/3/29=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C011/Q:0
S:80/1/2A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BA35/Q:1
S:80/3/2B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =E87A/Q:0
Rules:0000000100=
S:80/1/2C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C288/Q:1
S:80/3/2D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =90C7/Q:0

stuartpittaway, 2020-04-06:
Connect the controller to itself by connecting tx to Rx. You should now get replies and lots of ignored packet errors

gen-ben, 2020-04-06:
this what I get when Connecting the controller to itself :

Apply default config
Found pcf8574
Connecting to WIFI
Connecting to Wi-Fi...
mode : sta(50:02:91:46:a6:45)
                             add if0
                                    scandone
                                            state: 0 -> 2 (b0)
                                                              state: 2 -> 3 (0)
                                                                               state: 3 -> 5 (10)
                 add 0
                      aid 5
                           cnt

                               connected with XperiaXZ1, channel 11
                                                                   dhcp client start...
       S:80/1/1=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925/Q:2
R:80/1/1=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =925**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=
S:80/3/2=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7301/Q:1
R:80/3/2=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =7301**FAIL PROCESS REPLY**
Timing:60000ms
S:80/4/3=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CA8E/Q:0
R:80/4/3=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =CA8E**FAIL PROCESS REPLY**
Timing:60000ms
ip:192.168.43.87,mask:255.255.255.0,gw:192.168.43.147
                                                     Connected to Wi-Fi.
3. Connected IP:192.168.43.87
Requesting NTP from time.google.com
Got NTP time: 07:19:07 06/04/2020
S:80/1/4=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C5D6/Q:1
R:80/1/4=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C5D6**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/5=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9799/Q:0
R:80/3/5=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9799**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=
S:80/1/6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EDBD/Q:1
R:80/1/6=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EDBD**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BFF2/Q:0
R:80/3/7=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BFF2**FAIL PROCESS REPLY**
Timing:60000ms
pm open,type:2 0
                Rules:0000000100=
S:80/1/8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =34AC/Q:1
R:80/1/8=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =34AC**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =66E3/Q:0
R:80/3/9=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =66E3**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=
S:80/1/A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1CC7/Q:1
R:80/1/A=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1CC7**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4E88/Q:0
R:80/3/B=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4E88**FAIL PROCESS REPLY**
Timing:60000ms
S:80/1/C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =647A/Q:1
R:80/1/C=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =647A**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=
S:80/3/D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3635/Q:0
R:80/3/D=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =3635**FAIL PROCESS REPLY**
Timing:60000ms
S:80/1/E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4C11/Q:1
R:80/1/E=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =4C11**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1E5E/Q:0
R:80/3/F=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =1E5E**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=
S:80/1/10=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C679/Q:1
R:80/1/10=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C679**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/11=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9436/Q:0
R:80/3/11=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =9436**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=
S:80/1/12=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EE12/Q:1
R:80/1/12=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EE12**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/13=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BC5D/Q:0
R:80/3/13=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BC5D**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=
S:80/1/14=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =96AF/Q:1
R:80/1/14=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =96AF**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/15=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C4E0/Q:0
R:80/3/15=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =C4E0**FAIL PROCESS REPLY**
Timing:60000ms
S:80/1/16=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BEC4/Q:1
R:80/1/16=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =BEC4**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=
S:80/3/17=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EC8B/Q:0
R:80/3/17=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =EC8B**FAIL PROCESS REPLY**
Timing:60000ms
S:80/1/18=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =67D5/Q:1
R:80/1/18=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =67D5**FAIL PROCESS REPLY**
Timing:60000ms
S:80/3/19=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =359A/Q:0
R:80/3/19=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0  =359A**FAIL PROCESS REPLY**
Timing:60000ms
Rules:0000000100=

stuartpittaway, 2020-04-06:
So the controller is working, its sending and receiving a reply.

Fault must be in the modules somewhere.

stuartpittaway, 2020-04-06:
Why do you need so many JST connectors if its all on one board - why have the TX and RX JST connectors ?

gen-ben, 2020-04-06:
JST connectors Normally not used but I thought that if one of the model's breakdown I can skip it using cable. just as temporary solution.
is there any way to debug the modules using serial port?.
How can I use "ATTINYCellModuleTestPrg"  i have uploaded the code all three LEDs start blinking and bypass starts.

stuartpittaway, 2020-04-06:
The ATTINYCellModuleTestPrg just tests bypass and the LEDs are connected properly.

I think the issue lies in the optoisolators and the comms between the boards, not much to debug unless you write your own code - the memory on the ATTINY is too small to add much debug code.

gen-ben, 2020-04-06:
is  Serial1 use PA1 = PIN 12 SERIAL TRANSMIT (TXD0)  and PA2 = PIN 11 SERIAL RECEIVE (RXD0)?

Serial1.begin(38400, SERIAL_8N1);
Serial1.println(F("DIYBMS Test Program"));

stuartpittaway, 2020-04-06:
help files for ATTINY core...

https://github.com/SpenceKonde/ATTinyCore/blob/master/avr/extras/ATtiny_x41.md

gen-ben, 2020-04-06:
when i am trying to read from serial ports "both  of them" in cell module I am getting strange character I couldn't manage to read from it also I couldn't find the reason.

the serial monitor look like this:

▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒

stuartpittaway, 2020-04-06:
That's probably a non printable character.  I can't really help as I don't know what you are looking at or what software you were writing!

gen-ben, 2020-04-06:
I only printed this : 

Serial1.begin(38400, SERIAL_8N1);
Serial1.println(F("DIYBMS Test Program1"));
Serial.begin(38400, SERIAL_8N1);
Serial.println(F("DIYBMS Test Program"));

just to check if the serial port working or not.
I have add it to ATTINYCellModuleTestPrg file.

stuartpittaway, 2020-04-06:
But that's going to be spitting it out to the optoisolator, so how are you going to read it, did you tap into the pins on the chip?

gen-ben, 2020-04-06:
I have connected GND to GND and TX from 841 to RX in USB to serial converter.

stuartpittaway, 2020-04-06:
Which serial port are you trying to connect to?  You need to connect directly to the pins on the chip to see the output.

gen-ben, 2020-04-06:
yes i did that I have connected the cable directly to the pins on the chip. I tried  to connect it to pin number 8 for Serial1 and to pin number 12 for Serial0 and both gave the same result as i send to you, also I tried to remove the optoisolator to check if the problem caused by it but I got the same result again.

stuartpittaway, 2020-04-07:
Just a thought, you didn't get the ATTINY upside down when soldered to the board?  Is pin 1 aligned?

gen-ben, 2020-04-07:
yes sure pin 1 aligned, everything working on the board but serial, I have tried today testing the attiny on separate test board without connecting any component to it I have powered it from battery 3.9V and I checked the serial communication I got the same non readable characters. and now I have just tried an old t841 and uploaded the same code the serial worked without problem, but all new ones not.

stuartpittaway, 2020-04-07:
Did you burn fuses on new ones? Faulty batch of chips? Counterfeit?

gen-ben, 2020-04-07:
No I did not, how can I do that?
I uploaded the code from PlatformIO without doing any extra thing.

stuartpittaway, 2020-04-07:
There is a "Upload and set fuses" option in VSCODE and PLATFORMIO

gen-ben, 2020-04-07:
I got this Erorr:
Error: Dynamic fuses generation for attiny841 is not supported. Please specify fuses in platformio.ini

stuartpittaway, 2020-04-07:
The fuses are in platformio.ini! 

```
; lfuse = 1110 0010 = CKDIV8 & Calibrated Internal 8MHz Oscillator
; hfuse = 1101 0110 = EESAVE & 1.8V BOD detection level
; efuse = 1111 0100 = ULPOSCSEL2/ULPOSCSEL1/ULPOSCSEL0/BODPD1/BODPD0
board_fuses.lfuse = 0b11100010
board_fuses.hfuse = 0b11010110
board_fuses.efuse = 0b11110100
;-B16 option needed for my USBASP programmer to slow it down!
upload_flags =
  -vv
  -B16
  -Pusb
  -Ulfuse:w:0b11100010:m
  -Uhfuse:w:0b11010110:m
  -Uefuse:w:0b11110100:m

```

gen-ben, 2020-04-07:
Thank you very much for your time and help. I burden you with questions, The problem was in the platformio.ini file. I am using Arduino UNO to upload the code and when I have changed the setting in platformio.ini file some how I removed fuses specification without realize. after adding it I uploaded the code normally and every thing starts working :) .

stuartpittaway, 2020-04-07:
Well we got there in the end!

One of the reasons for the modular approach in DIYBMS is the danger of short circuiting a large battery bank - please be very careful with the board you have made, there is very little separation from one module to another!

gen-ben, 2020-04-07:
I will keep an eye on it and test it for a while before connecting it to my battery bank. thank for the advice and I wish you a nice day.

closed 2020-04-07 by gen-ben