stuartpittaway/diyBMSv4Code #12: MQTT Features

nick81, 2020-04-05:
Hi Stuart,

Can you extend the MQTT messages for the whole pack/bank? 
- pack voltage
- relay status
- voltage range

stuartpittaway, 2020-04-07:
Hi Nick, do you need the pack voltage and range?  Can these not be calculated in what ever system is processing the MQTT messages?

nick81, 2020-04-20:
Hi Stuart, sorry for my delay. 
Sure, I clould calculate the pack voltage. It only would be easier to get the infomation directly. MQTT is a lightweight protocol, so i thougt, it would be a nice option. The active rules / relay status would be nice too. Or do you have performance issues with the wemos?

nick81, 2020-04-23:
Hi Stuart,
as additionally opinion. I read about mqtt length limitation on the forum.
https://community.openenergymonitor.org/t/diybms-v4/11292/1318
Why do not publish multiple mqtt messages? e.g. one for each bank and one for global information (status ralay, etc)

MaartenGoris, 2020-04-25:
nick,

i have a tested version that send information per bank. because there was a limitation in the TCP buffer. need to do a Pull request but not sure on which branch.

@stuartpittaway  Can you help me out.

Adding more bank information is something that is possible if ok for Stuart, i was also thinking 
to subscribe the controller in diybms/request topic to request the bank status, this can provide 
a nice view when refreshing a page for example. now the timer is 30sec.

MaartenGoris, 2020-04-26:
@nick81 @stuartpittaway 

I added the feature to be able to request a bank status, 
`diybms/request` with the bankId as raw data will update the requested bank.
Also the total voltage is foreseen, 
I have foreseen a relay status, but this is not yet filled out.
Working on that.

[image]

nick81, 2020-04-26:
Looks great 👍

stuartpittaway, 2020-04-28:
Maarten what overhead does the MQTT requests place on the ESP8266?

I think keeping with a push mechanism would be less burden on the controller.

MaartenGoris, 2020-04-29:
@stuartpittaway Any idea how to test that?
I was thinking to record the cycle time if not subscribed to the topic 
en compare this with subribed cycle time. 

I can keep this update out, but i would like to implement the sending per bank,
GIT is new for me and cant figure out where to do the pull request

stuartpittaway, 2020-04-29:
If you clone the whole diyBMSv4Code repository  to your own account make the changes there and then submit a pull request I can then merge the changes you make into my master version

daromer2, 2020-12-06:
Generally push is enough and above data could be pushed every 5-10 seconds without significant issues. I have multiple esp8266 doing 100x the work that current code does in terms of sending mqtt without issues. 

I would say we should just add this data to mqtt and send them out. Perhaps in near future under the mqtt tab add like 3 options in terms of:
* sending cell information
* sending overall bank information
* sending other? 

It might be that you in near future might even want to have on-demand-push like for instant sending notification of trigger changes and such?

Michaelillingby, 2021-01-06:
how do I connect to the MQTT server when my server doesn't require a username or password? it is just isolated on my LAN and therefor no need for authentication.

stuartpittaway, 2021-01-21:
The updated MQTT code is now in master branch with the changes @daromer2 made.

closed 2021-01-21 by stuartpittaway