stuartpittaway/diyBMSv4Code #13: Relay gets activated for a short time when ESP board restarts

wurstyboy, 2020-04-08:
I use the 4 channel relay board as recommended.
On relay channel 2 I have two rules active:
- Individual cell under voltage: 2900, on
- Pack under voltage: 42000, on
Relay default is: off

Some times my esp controller restarts on its own. I guess some sort of overflow (many packets get los and I also have some CRC errors) or in generell some noise problem. But anyway, after the controller restarts the relay 2 gets fired up for some short time. Thus, my battery inverter gets the open circuit signal and thinks that my pack is critical low. As consequence it starts charging (from the grid). The same happens if I restart the esp on purpose (by software and by unplugin/plugin).

Maybe the esp controller doesn't calculate the total pack voltage in the first secondes correctly as it doesn't have the feedback of all modules. If this is true we need some sort of delay for the relays.

I tried changing to relay channel 3 with the same setting and the same issue happens. So I think it is not a relay board based problem.

stuartpittaway, 2020-04-08:
Thanks for reading the issue.

When the esp is rebooted/reset the output pins will also change state to low or high. At this point they will drive the relays to the state of the pin.

I can't control this in code because at this point my code isn't running the esp is in a boot up reset process.

Your esp shouldn't be rebooting though! How many modules are you using?

wurstyboy, 2020-04-08:
I have a 14 s li-ion setup, thus 14 modules. The wires from module to module are 20 cm each. With the old code (like 1 month ago) the ESP never was restarting. But I updated because of the better packets sending code. I have a lot of noise in the system. As soon as the charge current exceeds ~10-20 A not a single packet gets through. But the restarts happend at night time during standby almost (just 2-5 A for my base load in the house).

I don't understand why only relay channel 2 gets a short high signal. I also have over voltage and over temp. rules on channel 1. But the relay 1 stays off during a reboot.

daromer2, 2020-11-30:
Stuart: Could we utilize an enable function on the relays? Since the ESP do some toogle work on some ports i guess we can use one that always is low to then pull and enable those outputs later on?  (Though note i havent studied the diagram that much so perhaps disregard my comment) 

But is Relay 2the one on P0 on the 8574p? On another thought should they really flicker at all since what i can see all are on the OI expander and the I2C bus should be pretty unaffected during bootup?

stuartpittaway, 2020-12-01:
I think the issue with the relays are when the controller is rebooted, the pins drop to LOW as the ESP initialises the ports on the expander.  The relay board is active low.

What we have seen is clone ESP8266 modules with low spec AMS1117 chips - they are not able to supply the necessary power to the ESP and they cause reboots.  This is most likely when the ESP is in a low WIFI signal area and is using extra power to "hunt" the router.  We've also seen ESP devices rebooting like this when the internet/wifi signal goes down/switched off.

closed 2021-01-21 by stuartpittaway