stuartpittaway/diyBMSv4Code #15: Clarify allowed use cases

adlerweb, 2020-05-07:
Hi,

sad to see you moved away from open licenses and are now using a more restrictive one - especially since "non-commercial" isn't really a well defined concept and a lot of uses might be considered "commercial". As such things like paying companies to fabricate boards instead of etching them at home, using boards to run a solar system connected to the grid (some countries require owners to start a "company" to get the necessary permits) or even talking about the system on websites containing ads might put users into a position where lawyers might write nasty letters. It would be great if you could add a little clarification on where your interpretation of "NC" starts and what uses are explicitly allowed.

stuartpittaway, 2020-05-15:
Hello adlerweb.  

Which particular points of the licence are you objecting to?

The open licence allows all non-commercial use.  So a DIY hobbyist has no limitations to adapt and modify the code.

adlerweb, 2020-05-15:
Hello,

I am not objecting to any part, I just stumbled upon the project while searching for current implementations and wanted to understand your intended uses. I used the term "away from open licenses" since CC-BY-NC is not considered a "open license" by the [OSHW](https://freedomdefined.org/OSHW#Open_Source_Hardware_.28OSHW.29_Definition_1.0) or the [OS Definition](https://opensource.org/docs/osd), is incompatible with most licenses and as such prevents me from taking ideas found here and mixing them with ones found in other projects.

As I noted: there is no clear definition of "commercial", the transition between "DIY" and "commercial" is pretty fluid and a lot of people have different ideas what should be considered commercial, so it's not clear to me which uses you intended to allow or deny by choosing NC. Some people see "NC" as _"you can do everything as long as you do not directly sell my boards in mass quantities"_, some as _"using it commercially is fine if you buy them from me"_ others as _"please do not engage in anything where someone could directly or indirectly make money by using/showing/etc my boards"_. I already posted some examples, here with a bit more explanation:

- **Connect the boards loaded with your software to a solar system** - as stated: I am legally required to start a "company" to get a permit to install a solar system. I also could technically "sell" electricity produced by my solar panels to the grid. Even tho I neither do nor intend to make money from it this is most likely considered as "used in a commercial environment". Since the license only states "commercial", without any explanation if this includes commercial use or only providing commercial services to others, this could be considered a not permitted use.
- **Use a board-service** - Nowadays I rarely etch boards myself. I contract a boardhouse, they take YOUR design, produce the boards and possibly even assemble it - and I pay them for it. So they sell your finished product to me. For me selling finished board is clearly a commercial thing. Still there are a lot of people using JLCPCB, etc to get diyBMS-Boards and I haven't seen any complaints, so I guess that's something you don't consider commercial? Maybe because it's not in their catalouge but only build on a customers request?
- **Buy as a group** - Usually it is cheaper to produce boards or buy components in larger quantities. If the project is interesting I usually ask arround in my local hackerspace or other interested groups if someone also wants to build it and "join forces". 5 Boards for me, 5 for you, and we both pay 50%. No profit besides maybe rounding odd amounts. Cheaper than everyone buying 5 themselves. Legally that's however considered a sale, so commercial, or is it?
- **Talk about the hardware/software on commercial platforms** I found diyBMS over videos on YouTube, similarly I'm active over there showing projects I am using or considered. In both cases YouTube, and most likely also the content creator, will get payed by ad revenues for showing your boards/software. Same goes for asking help on a online forum - most got ads, so one could argue the forum operator makes money by using your creations. Money flows, so posting there could be considered commercial.

That's what I ment by "no clear definition" and why I suggested to document a few use cases to get possible users an idea of what you consider allowed and what not.

stuartpittaway, 2020-05-15:
I see what you mean, from my perspective I don't permit people to profit from my design.  In that I don't wish to see a shop selling them at a profit without my permission/agreement.

People driving revenue from the design by other means such as indirect advertising on YouTube or Google Ads on a forum for instance is unavoidable.

closed 2020-11-02 by stuartpittaway