stuartpittaway/diyBMSv4Code #24: SAVE D3 pin (for reset config) and a delay (3 seconds) at start up

ortegafernando, 2020-06-22:
Hi Stuart, in order to save one pin in wemos, you can use double reset detector library instead of D3 pin.

After deleting Wifi settings by pushing D3 at boot during a 3 seconds delay and with the "risky" of bounce buttons you finally restart ESP8266.

So there is no problem to use doble reset detector library, it resets the ESP8266, but is the same, finally you get what you want by only by pusing twice reset button. After detecting that, you can run the SoftAP module.

I hope you can implement this in future, a pin in an ESP8266 is like petrol !!!

ESP8266 version: [https://github.com/datacute/DoubleResetDetector](https://github.com/datacute/DoubleResetDetector)
(I have used it in several projects, works fine)

ESP8266 and ESP32 "new version for another guy": [https://github.com/khoih-prog/ESP_DoubleResetDetector](https://github.com/khoih-prog/ESP_DoubleResetDetector)
(I have never use it with ESP32, but may you you prefer it for future migration to ESP32)
It even has an example of how to use it with WifiManager library. Think also that wifimanager will be ready for ESP32 soon, (it is almost). [https://github.com/tzapu/WiFiManager/tree/development](https://github.com/tzapu/WiFiManager/tree/development)

Bye.

stuartpittaway, 2020-06-23:
Thats an interesting idea, thanks for that, will take a look.

stuartpittaway, 2021-01-21:
I'm not going to fix this in the current controller, but have in the next version of the PCB.

closed 2021-01-21 by stuartpittaway