stuartpittaway/diyBMSv4Code #25: Bank zero disappears periodically and loses max and min values when reappears.

hotear, 2020-06-24:
I have 3 banks (14s) attached to my controller.  When watching the monitor page from the web browser, periodically (when charging mostly) bank 0 will disappear from the display but bank 1 and 2 will remain.  On the next refresh it will reappear but it will have lost it's min and max voltage values.  I've attached a couple of screen shots showing the data missing from the monitor.json file and when a bank is missing (not that banks are not renumbered, bank 1 and 2 remain as 1 and 2 not changed to 0 and 1).
[image]
[image]

hotear, 2020-06-24:
I think I know where it's happening in the code and I have an idea for a fix but what I don't yet know is why it is happening.

stuartpittaway, 2020-06-24:
What version of the code are you running?  I think this has already been fixed and relates to a JSON buffer being too small.

hotear, 2020-06-24:
I don't think this is the json buffer (though I could be wrong), as I'd expect the last bank to disappear rather than the first.  I think the issue is from line 91 of PacketReceiveProcessor.cpp.  The number of modules is != previous number of modules so it resets the max and min.  I'm going to try a fix a bit later that will ignore a difference in modules unless it occurs for more than a couple of goes (as it only seems to last one update).  What I'm not sure about is why the number of modules in the first bank comes back as less than 14.

stuartpittaway, 2020-06-24:
The screen shot shows ignored request errors this should always be zero, it basically means that all the modules ignored a request from the controller.  Is this related?

hotear, 2020-06-24:
Not sure at this stage.  Under what circumstances would a module ignore a request?

stuartpittaway, 2020-06-24:
It should never ignore a request. The only time this could happen is if the modules get confused on their bank address

hotear, 2020-06-27:
So I think it is related to the Ignored requests (as you stated).
This section:
      } else {
        //Error count for a request that was not processed by any module
        totalNotProcessedErrors++;
        //Invalidate the whole bank if a module didn't process the request - something is a miss
        //or we have just configured a module to another bank
        numberOfModules[ReplyFromBank()]=0;
      }
Is what causes the bank to disappear.  This only happens if no module has processed the request.

I'm trying to understand all the situations that could cause no module to process the request.

hotear, 2020-06-27:
So the key is these 2 lines:
`      if (isPacketForMe()) {
        if (processPacket()) {
`
If I assume that the bank the cell module thinks it's in doesn't change.  Then the only way it seems that isPacketForMe could return false is if the address was 15 or 16 (no such module).  For processPacket to fail it seems that command would have to not exist (>6).  I wonder if the transmitted data is being corrupted in such a way that the CRC check is still valid (many bits have changed) but that the address or command have been changed in the way mentioned?

stuartpittaway, 2020-07-22:
I'm using a CRC32 so really unlikely that wire/line corruption would keep that valid.

stuartpittaway, 2020-07-28:
Did yo get any further with debugging this?  Have you tried the latest code

hotear, 2020-07-28:
Not had much time but was hoping to try something tomorrow evening. It's
quite time consuming because I have to detach all 14 modules from the
battery to reflash with the trial code.

hotear, 2020-07-30:
I had time to try something last night but I don't think it has worked (I'll know for sure later today when I start charging).  I'm going to switch around the order of my 3 banks to see if this problem is associated with a physical bank or just the logical position of the bank (i.e. it always happens to bank 0 regardless of which actual battery that is). Assuming it hasn't worked I will need to hook up a terminal and add some debug I think to really understand what is going on.

closed 2021-01-21 by stuartpittaway

hotear, 2021-02-15:
Sorry it has taken me so long to get round to updating to the latest version.  I finally managed to get it all done on Saturday (took me all day).  I can report that so far the problem seems to have been fixed and bank 0 no longer disappears.

stuartpittaway, 2021-02-16:
Thanks for updating!