stuartpittaway/diyBMSv4Code #27: Cell modules moved away from bank 0 don't register over voltage

Fred7421, 2020-07-23:
Any cell modules moved away from bank 0 do not initiate balance mode. Red LED does not light and resistors do not heat up. This happened on banks 1, 2, and 3. I lowered global bypass voltage threshold below my lowest cell, save settings and cells only on bank 0 did what they should.

stuartpittaway, 2020-07-23:
Thanks for this report.  I think this has already been fixed in the next version.  The "global" button only applies to bank zero but should set all banks.

You can manually change the balancing settings for each cell, but have to do each one individually.

stuartpittaway, 2020-07-24:
The code that fixes this issue is now available in the master branch.

closed 2020-07-24 by stuartpittaway