stuartpittaway/diyBMSv4Code #29: A Relay Wishlist

Fred7421, 2020-07-24:
See attached image. I figured a pic is worth a thousand words, in this instance.
[image]

stuartpittaway, 2020-07-28:
Thanks for taking the time to write this up!

Is this just a different screen layout or do you expect the rules to operate differently as well?

What priority is given to the rules/sequence ?

Fred7421, 2020-07-28:
I figure I should give back a little if I can. Plus added versatility would be nice.

As the rules stand now, you can only associate the settings for one or all of the relays. Each relay is not the captain of its own ship. If I wanted to have relay 3 turn on at 3pm and relay 2 turn on at 2pm, I can't. (All the relays are on the same ship).

I'm working on a mock python3 code (within my abilities) for further clarification. (do I post it here?)

As far as the priority is concerned, I was thinking... 'if this == True, or this == True, or not 'X' (if 'X', turn all off) then relay state = not relay state.

Basically all choices would affect the relay state using an 'or'. So priority would live at the Global variable level (within the 'Rules' universe). Like: which relay are we talking about? What type of relay is it? Is the relay being used? if so, is it NO or NC? and do you want it to flash? etc. Those would be the foundation (singular) to which all other settings would act on. 
If one or more settings is True, flip the switch.

Maybe frequency of events is a factor. Emergency stop; not very often. Comms Error; probably a little more often than emergency. Over/Under voltage/temp more often than comms error, etc. 
It may not matter. If multiple rules end up being True at the same time, it still flips the switch.

stuartpittaway, 2020-07-29:
Thanks for leaving more detail.  I under stand the problem of the timer example you give, as currently there is only two rules for that particular element.

The existing rules are also in priority order (in what I thought was a sensible criticality).  I'm not saying this is right or wrong, but how would that also work in your example?

For instance, over heating or over voltage are a higher priority than a time rule

Fred7421, 2020-07-29:
I guess most of my wish list is re-inventing the wheel. The current rules/relay choices cover a lot of what I was thinking.
I was going for a 'per relay' type of configuration. Have the ability to customize each relay's on/off setup
If there's an execution order/sequence, using if/else could work. If no 'comms error'; move down to the next 'if' statement, else: switch its state type of thing.
Put the highest priority first in line.

stuartpittaway, 2021-01-21:
Just to mention that module temperature monitoring is in the latest release