stuartpittaway/diyBMSv4Code #30: WARNING! Cell modules overheating, PID crashed

davidbue, 2020-08-15:
Dear all,

I have been running with the DIYBMS for 3 weeks now. Running flawless, till I did a configuration change that crashed every single one of my 14 cell modules which began to go smoking hot.

Running the current release version of the code. 

I wanted to balance my power wall down to 3.8v/cell to do some charge testing with my solar tomorrow. The stack was at 3.9v/cell. Here is what happened. 

1) I updated the global settings to 65 Celsius, 3800 mV,
2) drank a cup of coffee, 
3) went down to my power wall, and noticed that the balance indicator LEDs were all solid on. Not “PID flickering”
4) pulled up the web interface. Each module was at 120 Celsius and rising.
5) in a panic, increased the bypass voltage to 4100 mV and did a global update.
6) all cell modules cut out now that the bypass voltage was set higher than what the cell voltage. The boards began to cool. LED went off for every single module.

This was reproducible. Each time I changed back the bypass voltage to any value below the cell voltage, the PID did not kick in. The LEDs just went solid on and the dump resistors was driven constantly on.

Everything else appeared normal. I could read out the configuration of each module. No bogus values, no bad data packages.

It was first when I physically unplugged and thus rebooted each cell module that I could get rid of the dangerous and odd behavior. After having cycled power to all 14 modules I could set the global parameters to 65 Celsius and 3800 mV and all cell modules have since operates as intended.

I do not dare to keep the system on without active supervision now. Has anything like this been seen/reported before?

PS. I am using the 4mb WEMOS on the ESPController. Do not know if this can be related to my issues. (I have a 16mb WE OS laying about but has been unable to get the code to run on it)

stuartpittaway, 2020-08-15:
Hello, obviously this is very concerning.

Are you running the latest module code on each of the modules? The latest code did change the PID vales, however on all the tests I've done over the last 6 months it's been flawless.

What version of the PCB are you using? Is it a 4.21?

The 4mb wemos won't make a difference.

davidbue, 2020-08-15:
I am running the master branch, downloaded 4th of July. To my knowledge, even though I have downloaded the master branch, you have not updated the cell module code for many months so it should be the exact same code as what is there now.

I have not made changes to the code.

I am using version 4.2, according to the footer in the schematic it's dated to 2020-01-25

stuartpittaway, 2020-08-15:
Due to the way GitHub branches work, the code has indeed changed on the modules.

Did you not get any ignored packet errors on the dark theme web pages?

On v4.20 boards there is a layout mismatch, the thermistor is in the wrong place. Did you swap the resistor? 

Are you using the correct programming option from platformio (see the YouTube programming video for more info)

davidbue, 2020-08-15:
I think that I was a bit off with respect to versions. I can see that I wrote in my initial post that I was running the current release. Not correct.

I am running the master branch of the code (both on the controller and cell boards) as downloaded on July 4th. So prior to the dark theme.

I just opened up KiCad in board. It is 4.21 according to the silk screen.Seems as the date code in the schematic footer has not been updated.

davidbue, 2020-08-15:
I could try downloading the July 2020 release and update. Hope that whatever glitch it might be, is gone.

davidbue, 2020-08-15:
Or if you want to know what happened as much as I do, then let me know what I can do to investigate further.

stuartpittaway, 2020-08-15:
Ok as long as the silkscreen version says 4.21 on the modules that's the latest version.

Please follow the video and use platformio to flash the latest code.

Please test the modules function as you expect, just in case there is a hardware fault or the thermistors are faulty.

When balancing starts the red led should be fully on, when the temperature gets to 10 degrees below the set point, the PID kicks in and should begin to flash the red led.

The web page should also report a 0 to 100% discharge level on the graph.

Let me know how you get on, and I'll look to add on a failsafe code for over temperature.

davidbue, 2020-08-15:
Dear Stuart,

Code updated to the 20200724 release with black theme.

All cell boards are running just fine and I replaced the WEMOS D1 from the 4mb version to the 16mb version.

I instantly noticed that the cell boards are behaving differently from the old code. Instead of the rapid flashing as the PID was working, I now see a much more slowly updating "on/off" behaviour. Guess it's still PID but with a much slower loop speed?

Anyways. I'll report back here if the issue re-emerge. If you don't hear from me the next few weeks, consider it gone. May have been caused by something in the old code.

Thank you for everything you are bringing to all of us. It is truly magnificent.

Cheers,
  David

davidbue, 2020-08-15:
One last question. How can one donate to your work? Paypal?

davidbue, 2020-08-19:
Time to report back with a status. After the update to release version 20200724 the boards seem to run stable. 

One thing I have noticed, and what I cannot identify the obvious cause for, is that the cell boards are being run, 10C below the chosen setpoint and I don't notice the PWM rapid modulation of the dump load.

I think this might relate to this part of the code, and maybe it is what has also been documented in issue [31](https://github.com/stuartpittaway/diyBMSv4Code/issues/31):

`if (bypassCountDown > 0)
  {

    uint8_t exttemp = PP.InternalTemperature() & 0xFF;

    if (exttemp < (myConfig.BypassOverTempShutdown - 10))
    {
      //Full power if we are nowhere near the setpoint (more than 10 degrees C away)
      hardware.DumpLoadOn();
      hardware.StopTimer2();
      PP.PWMValue = 100;
      pwmrunning = false;
    }
    else
    {
      if (!pwmrunning)
      {
        //We have approached the set point, enable PWM
        hardware.DumpLoadOff();
        //Start timer2 with zero value
        hardware.StartTimer2();
        pwmrunning = true;
        //myPID.clear();
      }

      //Compare the real temperature against max setpoint, we want the PID to keep at this temperature
      PP.PWMValue = myPID.step(myConfig.BypassOverTempShutdown, exttemp);

      //Scale PWM up to 0-10000
      hardware.SetTimer2Value(PP.PWMValue * 100);
    }
`

stuartpittaway, 2020-08-19:
Thanks for the feedback.  Can you try the code from this branch?

https://github.com/stuartpittaway/diyBMSv4Code/tree/ModuleCodeImprovement

I've made some minor amends to improve the PWM function.

davidbue, 2020-08-19:
I'll see if I can update the code tonight and report back the results. Will that be only the cell boards that needs updating?

stuartpittaway, 2020-08-19:
Yes, just the modules.  Feel free to just do a few to test the change rather than the whole bank.  Will then give you something to compare against.

davidbue, 2020-08-19:
Roger that, will do.

stuartpittaway, 2020-08-19:
I think I managed to get a module to go over temperature earlier.  It seemed to happen when a module was disconnected from the controller whilst it was in the middle of balancing (red light on).

The PWM function didn't appear to kick in at all in this scenario and the temperature went over 90 degrees C.

stuartpittaway, 2020-08-20:
Hi David, not sure if you had chance to try the code.  I've added in lots of safety checks and other improvements to the code.

Unfortunately I've still managed to get 1 module over the limits, and at the moment its unexplained.  I've never seen this behaviour before - in months of testing!

I wondered originally if the ATTINY had hung/crashed, however its still replying to the controller and the watchdog doesn't kick in.

I'm now wondering if the MOSFET is actually stuck "on", so the ATTINY thinks its stopped but it hasn't.

davidbue, 2020-08-20:
Hi Stuart,

I arrived home late yesterday from a long evening at work, trying to clear up noise on a CT scan that I had a hard deadline on and thus did not get around to test it. It is my plan to do so tonight.

The behavior you describe is very much like what I saw. The ATTiny did not lock up. All cell boards still reported back when queried for their configuration.

There is a 47k pull-down on the mosfet so i would be surprised if it gets stuck open. Also because I can see that pin 10 on the ATTiny only drives the mosfet and no other fuzz is going on. It would really surprise me. I’ll hook an oscilloscope up tonight to see how hard the gate is being driven high/low.

I propose a test. How about updating the code such that the blue LED on pin 8 is set to the same state as pin 10 (the dump load enable pin) whenever the dump load is turned on/off. Then, if the blue LED gets stuck in an on-position it must be in software.

davidbue, 2020-08-20:
Also, what did you do to provoke the issue? It would be great if I can actively attempt to trip the cell board.

stuartpittaway, 2020-08-20:
I've no idea what provoked the issue!

I'd flashed most of the modules with the newest code, and then forced a "global" settings change to lower the voltage so they would all attempt to balance.  At that point a couple (not all) of them did as described.

Once I pulled the power on the module and rebooted it, all fine and I've not been able to recreate since.

I did add a capacitor about a year ago as the ATTINY was randomly rebooting when the MOSFET power surge kicked in, this fixed the issue, but just hope it didn't cause the MOSFET to not be driven correctly.  I've looked at the data sheet and it seems to be a capable part for what I'm trying to use it for.

davidbue, 2020-08-20:
Setting up the scope now. :-) Get back to you in a little while.

davidbue, 2020-08-20:
It is not the capacitor. The mostfet is driven nice and hard. Images incomming.

davidbue, 2020-08-20:
Sorry for the sloppiness of just taking photos of the scope screen. On the attached images you can see the gate voltage going low to high and going high to low. Both the rising and falling edge looks nice and crisp and the mosfet is driven all the way down.
[image]
[image]

davidbue, 2020-08-20:
Hi Stuart,

Final update. I am now running the "CodeImprovement" branch on the cell modules. Feedback:

1. Prior to the update, I had the issue with the PWM function not working properly with the July release of the code. Now, the cell modules are again rapidly flashing on the red LED, and the PID loop is working properly along with the transition from the _if (!pwmrunning)_ condition. **Excellent!!!**

2. This also means that the boards run at setpoint temperature and not 10C below. **Excellent!!!**

3. I cannot trip the cell modules. So on that matter, it's still inconclusive as to what caused the issue. I'm still slightly concerned about that. Could it maybe be caused by an overworked ATTINY that does not finish a code segment before an interrupt is fire'ed and it locks up? Maybe an overflow somewhere? I've skimmed though the code and don't see anything that could obviously make the code hang or overburden the MCU. I've just found this to be the cause of so many weird and unexplainable issues from previous projects where I thought I didn't push the MCU to its limits but I did so though an incredibly ridiculous block of code that did not look suspicious.

4. I am planning to run a potentially destructive test this weekend. I'll bring home a thermal imaging camera from work and image the board as I intentionally run it with the mosfet gate set constantly high from code. This may cause a critical meltdown of the entire cell module. Potentially even fire! Exciting! If this happens, we will have thermal imaging video of how bad the melt-down is.

stuartpittaway, 2020-08-20:
Everyone likes fire 👍 just not in your house!

With the amount of code checks I've put into the module I think it must be a hardware problem, but the scope pictures prove otherwise in "normal" operation.

The only other thing I've read about is excessive heat on the MOSFET causing it to lock on, but this wouldn't explain why a module just suddenly starts doing it.  Most report that once this has happened then the MOSFET is toast and is perm. locked on.

Perhaps heating the MOSFET is something to try?

stuartpittaway, 2020-08-20:
If you do manage to destroy a module, I've plenty more and I'll send you a couple if you don't have any spare!

davidbue, 2020-08-20:
The mosfet has the following characteristics:

Turn on delay time: 3ns
Turn on rise time: 2.5ns
Turn off delay time: 25ns
Turn off fall time: 4ns

I do not believe you are anywhere near switching it too fast. I can probe around and see if I can find anything suspicious on the hardware. I am though inclined to believe that it is in software. MCUs are so nice and versatile but at times they do really weird things.  :-D

I had 25 boards made from JLCPCB so I'm well supplied, but thanks. If I manage to capture some good destructive test video this weekend, we could consider making that test for each revision of your hardware, if you have one of each to spare. Would make a nice little compilation video for your youtube channel and I'd gladly carry out the test for you.

Cheers,
  David

stuartpittaway, 2020-08-28:
Hi David, did you make any more observations or release any magic smoke?

davidbue, 2020-09-07:
Morning Stuart,

I have not forgotten this. Work-wise things have just been a mess recently, will happen soon. Promise. :-)

davidbue, 2020-09-07:
Another mention as relates to this bug. I have tried powering the WEMOS from different sources. A standard Apple iPad AC USB charger, some different small Chinese buck step down dc-dc converter boards etc.

With most sources I get data package drops when my solar grid tie inverter kicks in. Any suggestions for a robust way to power the WEMOS  before I go noise hunting myself? 

On the plus side, none of the bad data packages has replicated the temperature runaway, spite that we could trip it by jiggling with the rx/tx wires.

martinbogo, 2020-11-11:
Stuart, I'm running 4.21 modules and the most recent JLCPCB code on the controller and the modules.  I am also seeing the modules running exactly 10C under the setpoint for the bypass on the graph.  My PWM is always pegged at 100% ( I never see the "fast flash" mode of normal PWM'ing ) so the resistor bank is working more like bang-bang than PWM.

This is a screenshot of my current setup -- I'm discharging to bottom-balance a 24V LiFeMnPo4 8S battery with a low voltage target of 2.5V.  For this example I set the bypass temperature to 80C to show the issue clearly.

[image]

stuartpittaway, 2020-11-12:
Hi @davidbue 

Thanks for posting a question.

When in discharge/balance mode the modules should be "fully on" (PWM not active) until the temperature of the module raised to 10 degrees C below the setpoint.  Then the PWM is started to take over and control the balance to keep the temperature of the module around the set point.

The screenshot posted, shows the modules at about 70C and the setpoint at 80C.  At this point are the RED LEDs fully on?

If you are running this in a cold climate or in a cold room, its possible that the resistors/module don't actually generate enough heat to enter the PWM phase so just run at 100% (which is good as the cells will drain faster).

At the voltage levels you are running, its only draining about 2W of power, so the modules may have enough heat sink capability to handle this.

If you set the bypass temperature to a low value - say 50C, does the PWM then kick in?

martinbogo, 2020-11-12:
Unfortunately, no... the pwm does not kick in at 50C.  Also, the modules never reach the set temperature... they always remain 10C below it.

This is using latest master branch as of yesterday.

stuartpittaway, 2020-11-12:
Thanks for testing @martinbogo.  What I don't understand is how the modules are not burning up if the PWM isn't kicking in, they should just get hotter and hotter.

Are you programming the V421 code using platformio ?  (and using the specific V421 option)

I've a couple of V421 boards on my desk, but they appear to behaving as designed using the code from master.

martinbogo, 2020-11-12:
I am programming them using PlatformIO, env version 421 on VsCode on MacOS.  The programmer is a usbasp.

screenshots of the interface, and a short video.. I configured them to 50C ... the modules only reach 40C and don't seem to be PWM'ing.

[image]

[image]

https://photos.app.goo.gl/EqnGibKWWUxMQPJj7

stuartpittaway, 2020-11-12:
Thanks for the video - that helps.

It looks like the modules are starting to balance, but then immediately stopping so the PWM never gets chance to kick in.

Does the BLUE LED ever light up?

Inside the loop() function, the code flow is similar to:

1. Turn on voltage reference (1.25V)
2. Take voltage and temperature readings
3. Switch reference off
4. Safety check the temperature is not over 90C - force shutdown if so
5. We then check the voltage (BypassCheck) if the voltage is higher than the setpoint (2500mV in your case) we enter a loop for 200 iterations
6. On every loop, we take a temperature and voltage reading, if the real temperature is less than "set temperature - 10C " then force 100% output to the resistors
7. Otherwise, start the PWM cycle to balance the output to maintain set temperature

After 200 cycles, we stop bypass and let the cell rest, if the voltage bounces back over the balance voltage, we start again.

stuartpittaway, 2020-11-12:
So, after all that, its possibly a hardware issue in that the voltage may be dropping below the brown out detector voltage (1.8V) so the module reboots when balance kicks in.

Can you check with a multimeter if the voltage drops a lot when the balance starts?  Check the voltage directly at the module power terminals.

karciok, 2020-11-12:
Hi, Stuart.
I have similar trouble with my modules. I saw that after first powerup of the battery module PWM works for couple seconds (resistor LED flashes quickly and module gets up to set temperature) but after like 2 or 3 loop iterations (turning the balancing off to measure voltage) PWM just doesn't kick in and module stays 10 +/-1 degrees colder than set temperature. The controller shows that PWM is at 100% and occasionally the bar goes to blue (not balancing) but after couple seconds it's red again. 
I don't think that it's voltage drop fault because module while balancing reports steady 50-80mV lower voltage than in not balancing mode.

martinbogo, 2020-11-12:
I checked with the multimeter and the voltage remains steady when the balance starts.  however it did lead to an experiment where I unplugged one of the modules.

When I plugged it back in it did one cycle where the pwm worked once!  Then it goes back to that Bang – Bang method where it doesn't do any pwm.

I think the timer that runs the pwm is failing somehow. 

I unplugged a module... here is a video of what it did:

https://photos.app.goo.gl/HqcPKJ4KtvyrgoMJ9

stuartpittaway, 2020-11-12:
Ok, looks like a bug then, can you try using code from a previous version from GitHub?

martinbogo, 2020-11-12:
Do you have a recommendation as to which commit I should go before?

martinbogo, 2020-11-12:
I think the issue may have snuck in with commit a5f58bd0ffed1e52c4fc01e80860ca99154ffd88

https://github.com/stuartpittaway/diyBMSv4Code/pull/47/commits/a5f58bd0ffed1e52c4fc01e80860ca99154ffd88

I'm not sure the PWM timers are being cleared correctly, or re-started correctly.

stuartpittaway, 2020-11-13:
Try the module code from this version (but ignore the controller code)

https://github.com/stuartpittaway/diyBMSv4Code/commit/504dfc986d76688ed9a2617ff93eb58d2c338f1d

stuartpittaway, 2020-11-13:
Interestingly, I've just deployed the MASTER branch code to a module (I was using V4.00 hardware module).  

I connected the module directly to a bench power supply, but not connected to a controller.  The module will operate in isolated mode and check voltage every 8 seconds.

At 4.1V the module began to balance, RED LED fully lit, pulling 0.5A, about 30 seconds later, the PWM started, current dropped to about 0.3A and the LED flickered as expected.

This cycle repeated until I dropped the voltage.

I'll try and test on V4.21 hardware.

martinbogo, 2020-11-13:
Sure.  Here are the module settings ... I set the bypass voltage to 2500 mV and bypass temp to 65C.

No joy.  This is also using latest master, freshly checked out, and a new install of VSCode + platformIO.

[image]

[image]

https://photos.app.goo.gl/bAkFUyWFfBtWX7jZ8

martinbogo, 2020-11-13:
I put the board under my microscope to verify the ATMEL is correct ( just in case there was a variant ) ... all other components are as specified for JLCPCB.  

[image]

My VSCode version is:

Version: 1.51.1
Commit: e5a624b788d92b8d34d1392e4c4d9789406efe8f
Date: 2020-11-11T01:11:34.018Z
Electron: 9.3.3
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Darwin x64 19.6.0

My PlatformIO version is:

Clove:ATTINYCellModule martinb$ pio upgrade
You're up-to-date!
PlatformIO 5.0.3 is currently the newest version available.

The build/upload output from the terminal is:

```
> Executing task in folder ATTINYCellModule: pio run --target upload --environment attiny841_V421 <

Processing attiny841_V421 (platform: atmelavr; board: attiny841; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/attiny841.html
PLATFORM: Atmel AVR (3.0.0) > Generic ATtiny841
HARDWARE: ATTINY841 8MHz, 512B RAM, 8KB Flash
PACKAGES: 
 - framework-arduino-avr-attiny 1.3.2 
 - tool-avrdude 1.60300.200527 (6.3.0) 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PacketSerial> 1.4.0+sha.840272c
|-- <FastPID> 1.3.1+sha.1792d23
|-- <settings>
|   |-- <crc16>
|   |-- <EEPROM> 2.0
|-- <Steinhart>
|-- <crc16>
Building in release mode
Checking size .pio/build/attiny841_V421/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=======   ]  70.7% (used 362 bytes from 512 bytes)
Flash: [==========]  98.8% (used 8092 bytes from 8192 bytes)
Configuring upload protocol...
AVAILABLE: usbasp
CURRENT: upload_protocol = usbasp
Looking for upload port...
Uploading .pio/build/attiny841_V421/firmware.hex

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/martinb/.platformio/packages/tool-avrdude/avrdude.conf"
         User configuration file is "/Users/martinb/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 16.0
[... 92 lines cut ...]
avrdude: verifying ...
avrdude: 1 bytes of efuse verified
avrdude: reading input file ".pio/build/attiny841_V421/firmware.hex"
avrdude: writing flash (8092 bytes):

Writing | ################################################## | 100% 6.92s

avrdude: 8092 bytes of flash written
avrdude: verifying flash memory against .pio/build/attiny841_V421/firmware.hex:
avrdude: load data flash data from input file .pio/build/attiny841_V421/firmware.hex:
avrdude: input file .pio/build/attiny841_V421/firmware.hex contains 8092 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 4.20s

avrdude: verifying ...
avrdude: 8092 bytes of flash verified

avrdude: safemode: lfuse reads as E2
avrdude: safemode: hfuse reads as D6
avrdude: safemode: efuse reads as F4
avrdude: safemode: Fuses OK (E:F4, H:D6, L:E2)

avrdude done.  Thank you
```

martinbogo, 2020-11-13:
_**SOLVED**_

This is a PlatformIO bug!  Fully uninstalling, then re-installing **and** upgrading PlatformIO to the latest version made the modules work.

I #@!?!@ hate undocumented fixes in tool chains.

Of course now this has to be added to the documentation ... because the failure mode is less than polite if you have a bad toolchain.

https://photos.app.goo.gl/Fj6rMZ5JhoAb2o9x8

stuartpittaway, 2020-11-13:
Thanks for the effort you put into this @martinbogo I was really scratching my head on this one - what a strange error!

stuartpittaway, 2021-01-21:
Latest code uses different PWM code and switches to Timer1 on the ATTINY.  Closing this issue for now, please reopen if problem reappears.

closed 2021-01-21 by stuartpittaway