stuartpittaway/diyBMSv4Code #34: Add hysteresis to relay rules

karciok, 2020-09-25:
Hi, 
I have been using your bms for a while and I noticed that when undervoltage or overvoltage events are triggered bms sends signal to turn off load/charging but as soon as device turns off voltage on the cells bounces back a little and bms sends signal to turn on the device. 

It would be great if You could add some sort of hysteresis, so when discharging the cells voltage reaches for example 3V/cell bms sends signal to turn off discharging device, but turning it back on would be possible when cells reach for example 3.2V/cell. Of course overvoltage situation working on the same principle.

It would eliminate constant turning on and off 
of peripheral devices.

stuartpittaway, 2020-09-26:
Hello, I've written the code for this feature over in a new branch, please take a look!

https://github.com/stuartpittaway/diyBMSv4Code/tree/HysteresisRules

Daemon256, 2020-10-02:
working!

stuartpittaway, 2020-10-07:
Hi Daemon256, is it still working as you expect?

Daemon256, 2020-10-09:
Now I will describe how it actually works (Firmware HysteresisRules, All rules X):
Relay default - OFF -> On the corresponding output, logic 1
Relay default - ON -> On the corresponding output, logic 0

And all the rules work out in exactly the same style. Logically correct to do the opposite.

Daemon256, 2020-10-09:
For correct work, you need to find the lines in the code:
RELAY_ON ? LOW : HIGH;
And replace them with:
RELAY_ON ? HIGH : LOW;

stuartpittaway, 2020-10-09:
That only applies to your configuration.  We probably need to add a new option for "inverted" outputs.

stuartpittaway, 2020-11-02:
Relay rules merged into master branch.

closed 2020-11-02 by stuartpittaway