stuartpittaway/diyBMSv4Code #35: Amount of Banks

bttfw, 2020-09-25:
Hi
Thanks for your work 

Is their a way to monitor more than 4 banks ?

Maybe 4 per controller and then a master controller to control the controllers?

bttfw, 2020-09-25:
I will be building it in a 2u rack case an will start with 5 but the end goal is 10-15 depending how my cell supply is stocked etc

I initially planned to use the watchmon 5 and multiple Multimons but yours is the same thing except for the price 

So I was wondering if I can get the same setup just diy

stuartpittaway, 2020-09-25:
Right.  At the moment it would be quite a change to alter the code (both controller and modules) to support more than 4 banks.

Others have asked for similar features, but I'm not sure the ESP8266 is up to it at the moment.

bttfw, 2020-09-25:
Yeah that's also a difference batrium uses an esp32 

I mean I have time just wanted to ask if this is planned in any way 

Have access to the code and make custom Pcbs is a huge plus point for your project

MasterCATZ, 2020-10-11:
same I rather build smaller banks , but having just a master controller for the controllers for users that need more banks would do the trick

should be less code changes as the controllers could keep functioning as they are

stuartpittaway, 2020-12-14:
The next code release supports up to 100 modules on the ESP8266 and current controller PCB design.

stuartpittaway, 2020-12-21:
Next release supports up to 16 banks and 100 modules on ESP8266

closed 2020-12-21 by stuartpittaway