stuartpittaway/diyBMSv4Code #37: Controller fails to send all module details over MQTT

stuartpittaway, 2020-10-01:
As reported in this forum post.
https://community.openenergymonitor.org/t/diybms-publishing-mqtt-to-emoncms/15557/13

ESP8266 seems to run out of power/steam after about 13 modules.  Look to split into lower packet sizes/batch sizes.

stuartpittaway, 2020-10-01:
Implemented changes to drip feed module data over MQTT.  Should we also do this on INFLUX DB?

Daemon256, 2020-10-02:
In the new version of the code - it worked

borpin, 2020-10-02:
You could do it as a single publish. Note the Inputs are topic_subtopic_sensor. If you built that as a single JSON payload, it would speed things up.

e.g.
{"0_0_voltage":4.5}
etc

Just watch for the payload character limit!!!

This reduces the load on emoncms and the MQTT Broker as well.

You could also reduce the name to V, Ti, Te, B to save characters.

closed 2020-10-27 by stuartpittaway