stuartpittaway/diyBMSv4Code #41: Current Sensing

DanielCh58, 2020-10-06:
Is there any sourcecode available for the current sensing? You you have any plans for that? Can I help?

stuartpittaway, 2020-10-06:
Hi Daniel, there isn't any source code available at the moment.  I'm just working on building an external current shunt for this exact purpose.

If you have coding skills available, then developing the controller code to have a "place holder" for the current (and external voltage) monitoring would be a good place to start.

DanielCh58, 2020-10-06:
I added a rotational resistance for faking different inputs on the sensing pin. So I can start developing something on my fork. If I have something useful, I will create a pull request.

[image]

stuartpittaway, 2020-10-06:
I doubt that I will use that analogue port, I'm looking at an rs485 interface, but it will be fine for prototype work.

DanielCh58, 2020-10-06:
I thought of a hall sensor: like [ACS712 30A Ampere Stromsensor Range Modul Hall-Effekt](https://www.amazon.de/dp/B08BZKPSFY)

DanielCh58, 2020-10-06:
> I doubt that I will use that analogue port, I'm looking at an rs485 interface, but it will be fine for prototype work.

But personally, I also prefer a digital solution. On the controller board we have no voltage ref. so the measurement is not so exact.

stuartpittaway, 2020-10-06:
I've got the ACS712 device on my desk now, they work fine, but for larger currents you need to move to a current shunt and a dedicated chip to read it.

jdevine82, 2020-10-23:
I too am looking to add current sense. I have used a Ina226 on other projects. I have only just built my BMS and am still programming etc. But once that is done I will be adding soc to this setup.

acidice333, 2020-11-20:
I'm also interested in this but I'm waiting for an ESP32 release of the controller before I order any modules and a controller.  A INA226 would be a good choice since its digital (i2c) and you can obtain "breakout" boards online for next to nothing.   You can change out the shunt resistor with a different one so you can go higher than 3A -- but the only downside is with this chip you're capped at 36v,

stuartpittaway, 2020-11-21:
The ESP32 release of the controller may be many months away, the existing controller board is fine and low cost.  The modules will be reusable on the new controller when available.

The BMS is designed for voltages up and over 48V, so I would need to include a current sensor that can cater for that.  At the moment I've been experimenting with the LTC2944 which goes up to 60V