stuartpittaway/diyBMSv4Code #44: Global Settings - Bypass threshold mV - does not work for other banks, except 0

Daemon256, 2020-10-25:
Global Settings - Bypass threshold mV - does not work for other banks, except 0

stuartpittaway, 2020-10-27:
Linked to #27

stuartpittaway, 2020-10-27:
I'm not sure why this code doesn't work as expected.  I sometimes find that it does and other times it needs more than 1 "save" of the global settings to make a difference.

```
void PacketRequestGenerator::sendSaveGlobalSetting(uint8_t totalNumberOfBanks, uint16_t BypassThresholdmV, uint8_t BypassOverTempShutdown)
{
  for (uint8_t bank = 0; bank < totalNumberOfBanks; bank++)
  {
    //Ask all modules to set bypass and temperature value
    setPacketAddress(true, bank, 0);
    //Command - WriteSettings
    _packetbuffer.command = COMMAND::WriteSettings;

    //Fill packet with 0xFFFF values - module ignores settings with this value
    for (uint8_t a = 0; a < maximum_cell_modules; a++)
    {
      _packetbuffer.moduledata[a] = 0xFFFF;
    }
    _packetbuffer.moduledata[6] = BypassOverTempShutdown;
    _packetbuffer.moduledata[7] = BypassThresholdmV;
    pushPacketToQueue();
  }

  clearSettingsForAllModules();
}
```

stuartpittaway, 2020-12-21:
Fixed in next release

closed 2020-12-21 by stuartpittaway