stuartpittaway/diyBMSv4Code #48: ESPController copilation issue

JoeMudr, 2020-11-05:
Hi, 
I'm trying to compile the ESPController software but am getting errors regarding git from the file buildscript_versioning.py.
When compiling from the downloaded Zip:
"fatal: Not a git repository (or any of the parent directories) .git"
When compiling from a cloned directory:
"fatal: your current branch 'master' does not have any commits yet"
Why does it want me to set up a git repository?

stuartpittaway, 2020-11-05:
The code was recently changed to automatically get the git version number at compile time as a number of people had issues knowing if they were running the latest version of the code.

Most people follow the compile video I put together which clones the git repo.

If you download the zip file this won't compile, just remove the line buildscript_versioning.py   in platformio.ini

JoeMudr, 2020-11-06:
Hello Stuart,
thanks for your reply. Removing the line didn't work. It got stuck on another line. I then tried your method from the video. that worked.

stuartpittaway, 2020-11-09:
Thanks for the feedback Joe, I've also amended the code to work with downloaded ZIP files.

closed 2020-11-09 by stuartpittaway