stuartpittaway/diyBMSv4Code #49: missing files when loading minipro

kenn1962, 2020-11-08:
Any input? Cloned the code per the video but only the 32 compiles, the minipro always fails.

:\Users\MadDog\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238:21: fatal error: algorithm: No such file or directory
 #include <algorithm>
                     ^
In file included from src\main.cpp:38:0:
C:\Users\MadDog\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238:21: fatal error: algorithm: No such file or directory
 #include <algorithm>
                     ^
compilation terminated.
compilation terminated.
*** [.pio\build\esp8266_d1minipro\libf31\ESP8266TrueRandom\ESP8266TrueRandom.cpp.o] Error 1
*** [.pio\build\esp8266_d1minipro\libb8f\ESP8266WiFi\BearSSLHelpers.cpp.o] Error 1
*** [.pio\build\esp8266_d1minipro\src\PacketRequestGenerator.cpp.o] Error 1
*** [.pio\build\esp8266_d1minipro\src\PacketReceiveProcessor.cpp.o] Error 1
*** [.pio\build\esp8266_d1minipro\src\main.cpp.o] Error 1
*** [.pio\build\esp8266_d1minipro\src\SoftAP.cpp.o] Error 1
*** [.pio\build\esp8266_d1minipro\src\DIYBMSServer.cpp.o] Error 1
xtensa-lx106-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\esp8266_d1minipro\lib0ae\Queue\cppQueue.cpp.o] Error 1
======================================================== [FAILED] Took 2.54 seconds ========================================================

Environment        Status    Duration
-----------------  --------  ------------
esp8266_d1minipro  FAILED    00:00:02.542
=================================================== 1 failed, 0 succeeded in 00:00:02.542 ===================================================
The terminal process "C:\Users\MadDog\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'esp8266_d1minipro'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

stuartpittaway, 2020-11-09:
Hello, the automated Travis build is passing without errors - log is here..

https://travis-ci.org/github/stuartpittaway/diyBMSv4Code/builds/741824874

The error you have seems strange - it seems to be reporting issues with the core Arduino.h libraries rather than an issue with DIYBMS code.  

Does this help?
https://community.platformio.org/t/arduino-h-file-not-found/11236

kenn1962, 2020-11-10:
Thanks Stuart for the reply. Read the link you sent but unfortunately most of it was way over my head. What I ended up doing was a clean install of visual studio and wiped all the directories for platformio before the reinstall. That seems to have done the trick. The build completed without error this time as did the upload. Odd thing is there was a file upload before further down the project task list like the video but isn't now . I can see the mini now as a wifi connection so that's a plus

daromer2, 2020-12-06:
@stuartpittaway  should be closed? :)

kenn1962, 2020-12-09:
Yes close it please

closed 2020-12-09 by stuartpittaway