stuartpittaway/diyBMSv4Code #50: Configure voltage range of graph and show nominal/high/low voltage for chemistries

martinbogo, 2020-11-11:
I work with cells that use different kinds of chemistry, and they have some voltage ranges that don't precisely fit into the "Lithium Ion" template of voltage ranges. e.g.)

- Lead Acid : 1.8 to 2.27 ( 2V nominal )
- LiFeMnPo (Valence) : 2.5 to 3.65-4.2V ( 3.3V nominal )
- LiPo : 3 to 4.2V hard cutoff ( 3.7V nominal ) 
- LiCoO2 : 3.0 to 4.2V hard cutoff ( 3.6V nominal )

As an enhancement, and since people don't tend to use mixed-chemistries ... I would like to propose an enhancement where the graph's voltage range fits the chemistry of the batteries being used and that the high/low/nominal voltages be shown across the graph as perhaps faint colored/dashed lines.   

When I am monitoring LiFeMnPo cells and (top and bottom) balancing them -- the graph really never extends more than about halfway up.  It's very hard to see small differences in charge between the cells when the voltage range is so small graphically.   

It's even more difficult with lead acid cells for example, since the voltage range on them is only 0.47V... everything is compressed into a tiny range.

[image]

[image]

stuartpittaway, 2020-11-12:
Hi @martinbogo 

Are you using mixed chemistry cells on the same controller?  Are these mixed cells connected together in a larger bank/pack?

The graph module/code I'm using is from eCharts (https://echarts.apache.org/en/index.html).

martinbogo, 2020-11-12:
I only use one chemistry at a time....but I have more than one power bank that I am configuring.

So my desire here is to show data appropriately for the cell type in use for a given monitor so that it is easier to see 'at a glance' what the overall state of charge/discharge without having to look at the axis scale.

I have one bank of lead acid powering a shed ... it is a 4S/20P system of marine deep cycle batteries.   Small voltage range means that the graph never scales quite right.

Another is acting as a UPS/Grid-Tie powerwall and is a 4KWH 16S48P LiFePo4 system.  

Each has it's own diyBMS v4 controller with a WeMos D1 Mini.

stuartpittaway, 2020-11-12:
Okay, that makes it easier then, I didn't know if you were mixing chemistry on the same controller.

It should be trivial to add on a min/max range to the scale and make that user selectable/configurable.

martinbogo, 2020-11-12:
Thank you!  Much appreciated.

stuartpittaway, 2020-12-05:
I've added this into the next version, in 256module branch

martinbogo, 2020-12-06:
Fantastic!  I'm happy and ready to give it a go.

stuartpittaway, 2020-12-21:
Fixed in next release

closed 2020-12-21 by stuartpittaway