stuartpittaway/diyBMSv4Code #52: Balance on demand

mkranjcic, 2020-11-16:
It would be nice to have option to make "force balance", for which you would enter mV value of biggest difference, and than modules would balance themselves regardless that cell is not in overvoltage state. 
This would help in case that you have one cell with lower voltage, and not getting them all to "charge state".
Thank you.

JoeMudr, 2020-11-17:
+1 for this. Maybe in combination with a current sensor: When there is no load or while charging, try to keep the cells within a certain Voltage range.

stuartpittaway, 2020-11-19:
How would you expect this to work, by "force balance" of the highest cell voltage to bring the average down ?

JoeMudr, 2020-11-19:
Something like this:

Balance Cell if:
CellVolt > LowCellVolt + BalanceHysteresis [ && Cellvolt > BalanceVoltTheshold]
[optional]
BalanceHysteresis & BalanceVoltThreshold can be set by user.

daromer2, 2020-12-01:
This function is often maintained from the master node. Due to that you in the end want to mix in the factor of current going in or out of the battery bank. Ie you generally only want to do this type of balancing when the battery pack is idle or close to idle.

daromer2, 2020-12-06:
Was thinking of this one on how we could work with it. 
Lets say the controller should master this. 
It need to be different from over voltage bypass so basically
while (!overvoltage) do
   if (this cell need to bleed)
       bleed(25 seconds)
  fi
done

so the maste har all the logic in its and it could potentially be
every 30 seconds
cellvolt > minimum balance voltage && diff(lovest_cell)  > hysteres && shuntcurrent (is between (max) && (min))
  send out flag to balance

Hmm. Just thinking out very loudly. Need to fit into current layout and design of the code.