stuartpittaway/diyBMSv4Code #53: Problems programming D1 Mini (no Pro)

JoeMudr, 2020-11-17:
Hi, 
I ran into some problems when programming my D1 Mini:
First i had to remove the line "upload_speed=921600". Otherwise the upload would fail.
Then after flashing i can connect to the Wifi but only see an empty page (http://192.168.4.1/softap.htm).
What's wrong here?

atanisoft, 2020-11-17:
@JoeMudr did you upload the filesystem after flashing?

atanisoft, 2020-11-17:
@stuartpittaway since you now have a gzip hook, have you considered embedding the compressed output directly into the binary so that the filesystem upload step is not necessary?

I was originally serving files from SPIFFS but found it to be unreliable and slow and so shifted to directly embedding via [this](https://github.com/atanisoft/ESP32CommandStation/blob/ac3af3f5d4263368b4c714b6c6110c2339dc06c0/build_index_header.py) script and used this block to serve it directly from the embedded blob:
```
  on("/index.html", HTTP_GET,
    [](AsyncWebServerRequest *request) {
      const char * htmlBuildTime = __DATE__ " " __TIME__;
      if (request->header("If-Modified-Since").equals(htmlBuildTime)) {
        request->send(STATUS_NOT_MODIFIED);
      } else {
        AsyncWebServerResponse *response = request->beginResponse_P(STATUS_OK, "text/html", indexHtmlGz, indexHtmlGz_size);
        response->addHeader("Content-Encoding", "gzip");
        response->addHeader("Last-Modified", htmlBuildTime);
        request->send(response);
      }
    });
```

Similar blobs can be created for other files you are serving.

stuartpittaway, 2020-11-17:
I used to do that, but find that doesn't work when compiling for esp32

atanisoft, 2020-11-17:
Any particular error? The above is from ESP32...

JoeMudr, 2020-11-17:
> @JoeMudr did you upload the filesystem after flashing?

After cloning the repository again, the option for uploading the filesystem ("platform"/...) under "env:esp8266_diminipro" was gone. Thought "Upload" would be sufficient then.
[image]
Clonded it again. And it's back. Now it's working. Thank's!

But still the fixed upload speed leads to an error.
"A fatal error occurred: Invalid head of packet (0x80)"

JoeMudr, 2020-11-17:
Alright. I got it up and running. BUT. I can't save anything. "Failed to save settings". Did I miss something?

stuartpittaway, 2020-11-17:
> Any particular error? The above is from ESP32...

Once the compiled code went over 1mb, it wouldn't upload. Although this was quite a while ago since I last tried it. I've recently swapped to littlefs and that does seem quicker.

stuartpittaway, 2020-11-17:
> Alright. I got it up and running. BUT. I can't save anything. "Failed to save settings". Did I miss something?

Just refresh the webpage. There is code to prevent cross site scripting attacks, if you reset the esp without refreshing the browser you will get that error.

JoeMudr, 2020-11-17:
> > Alright. I got it up and running. BUT. I can't save anything. "Failed to save settings". Did I miss something?
> 
> Just refresh the webpage. There is code to prevent cross site scripting attacks, if you reset the esp without refreshing the browser you will get that error.

Thank you. That helped.

atanisoft, 2020-11-17:
> Once the compiled code went over 1mb, it wouldn't upload.

Ahh, that is likely due to needing to adjust the partition definition. Probably will need a minor edit in the platformio.ini file to tell it to use a larger app instead of the default which has a large SPIFFS partition.

LittleFS works on the ESP32 as well *BUT* serving the content directly from flash (compiled into the app) will be considerably faster than SPIFFS or LittleFS.

stuartpittaway, 2020-11-18:
@joeMudr - are you all working now?

I think platform io has had a few bad upgrades lately and a number of users have seen strange behaviours.

stuartpittaway, 2020-11-18:
> @stuartpittaway since you now have a gzip hook, have you considered embedding the compressed output directly into the binary so that the filesystem upload step is not necessary?
> 

> Similar blobs can be created for other files you are serving.

Have you seen or tried this?

https://docs.platformio.org/en/latest/platforms/espressif32.html#embedding-binary-data

atanisoft, 2020-11-18:
@stuartpittaway I'm using that hook point for cmake based builds and generate a compressed gz version of the files using external tools. One benefit of the python script approach is there is no dependency on an external gzip utility though.

stuartpittaway, 2020-11-18:
Doesn't work on ESP8266 unfortunately, shame they don't port these features back

closed 2020-11-19 by stuartpittaway