stuartpittaway/diyBMSv4Code #55: Meassured voltage should not be done during balancing

daromer2, 2020-12-01:
Currently we read the voltage without looking into if the balancing is running. 
When balancing there is a significant voltage drop over the wires. Either we need to calculate this during voltage reading or we should just stop the balancing during this time

The drop itself do trigger the pwm to engage rather early. (I have to dig into the code a bit more to understand how it all works)

I could probably take a look but the code for the cell modules arent what im used to (Im not a coder) :)

stuartpittaway, 2020-12-01:
Hi, I used to ignore the voltage readings on v3 of the BMS, but felt that hid what was going on.  V4 does read the battery voltage during balance, perhaps I should just return the voltage at the point when balance starts.

Alternatively I do know what the resistor values are so I could just artificially increase the voltage to mimic the internal resistance of the circuit.

The drop shouldn't cause the PWM to trigger early, can you explain a bit more?

daromer2, 2020-12-01:
Start voltage
4.14V
Balance trigger at 4.1V.
During balance i can see it drop to belo 3.95V. And then just shortly after it stops balancing and bounces up towards 4.1+ again and just after it starts balancing again. Its not the temperature of the boards itself since they are around 40-45degree and the limit is still at 65. 

So by the look of it the voltage drop seem to cause the balance to happen. I can do some logging and do some graphs if that would help. I will also take a look at that code part later this week to see. 

But generally when balancing you stop the balance during checking the voltage levels.

daromer2, 2020-12-01:
What im aiming for is that as of now it stops due to "false" readings due to the voltage drop over the wire. If we meassure the voltage during the bypass quickly is paused this doesnt happen and you potentially gain some more balance time :)  With that said i guess thats not an argument enough but for me that pushes the data out i dont like to see the spikes with voltage going up and down during balance.

There is though one thing that it can affect. The fact that you might want to controll charging based on voltage readings. If the bypass causes the meassured voltage to be lower during those 30 seconds you cant rely on that the voltage is correct and thus you could end up on overcharging instead. 

Havent explored all options in current code but i push the data and controlls my chargers on my other bms system with this info :)

stuartpittaway, 2020-12-01:
The "bounce back" is the problem, once you stop balancing it can take several seconds for an 18650 to recover, so the voltage slowly drifts back up, so even if you do stop the balance when you take a reading, it won't be accurate.

I do agree that the voltage looks a bit odd when it bounces around during balance, and this could affect the rules placed on the voltage readings.

daromer2, 2020-12-01:
Yes but the natural bounce is just a fraction of the voltage drop on any decent size battery :)  We are not talking about balancing old 2Ah cells but 100+Ah cells/packs. They hardly bounce back anything with this small amount of current.

stuartpittaway, 2020-12-01:
Understood, my test bench is using 8x single cell 18650, so I've probably got the worst setup in that regard.

stuartpittaway, 2020-12-01:
Take a look at the latest code I've just pushed up, it switches off balance during serial comms traffic and also during balance.  Seems to return more stable voltage readings in the web interface

daromer2, 2020-12-01:
I will now. Where do you want reports for that branch? As normal issues?  For instance the integrity setting for the js file is wrong. 
[image]

stuartpittaway, 2020-12-01:
Mine looks ok, has it cached in the browser?

Do you want to report any issues as GitHub comments on the code directly?

daromer2, 2020-12-01:
Like if there is a way to connect an issue to a branch?

Updated the cellmodules with the new code but still drops around 0.2v. Though need to crosscheck if i didnt miss something when uploading. But i checked the code and looked ok hmm

stuartpittaway, 2020-12-01:
The voltage should drop though, you are pulling up to 1amp through tiny little wires so the resistance is going to cause a volt drop.

What size cell pack are you using?

daromer2, 2020-12-01:
Yes, but thats what we talked about ie shut down the bleed while taking the voltage meassurement. I understand that there is voltage drop over the wires itself. :)

The code looks fine so i need to crosscheck if the issue now is actual drop or me just done something else. Dont have more time to check today so will do some changes tomorrow and retest.

daromer2, 2020-12-04:
It looks allot better now. I need to do another test tomorrow and then i close this issue.

daromer2, 2020-12-06:
Closing for now. Its a lot better :)

closed 2020-12-06 by daromer2