stuartpittaway/diyBMSv4Code #56: Mqtt keeps broadcasting stale data when comms have issues.

daromer2, 2020-12-01:
What happens:
The current code keep sending same data even when comms is down or stale. 

Expected:
It should revert to 0 values if not valid data in them or just not send the data at all. 

Perhaps do something like this:  (main.cpp)
  if ((!mysettings.mqtt_enabled && !mqttClient.connected()) || receiveProc.HasCommsTimedOut())

stuartpittaway, 2020-12-05:
Made improvement to MQTT logic and also outputs relay and rule states

daromer2, 2020-12-05:
Nice. Closing this one

I have done some extra changes in mine. question is if and how you want to have such things added and in so in what way?  
What i have extra in my code right now is
* mAh logging on bypass
* Extra values sent over mqtt for the cells
* mqttStatus sent every 25 seconds containing generall status of whats going on. 

Can i ping you somewhere on perhaps talk about this. I cant see anywhere on how you want contributions to be done or/extra stuff. I have a list i think could be good to have :)

closed 2020-12-05 by daromer2

stuartpittaway, 2020-12-05:
Happy to chat, are you okay to drop me a private message on the openenergymonitor forum? Rather than sharing email addresses publicly.

mAh counting was something I was planning to add so glad you did that!