stuartpittaway/diyBMSv4Code #58: Can't compile 256Modules branch for esp8266 but works for ESP32

donnib, 2020-12-06:
I get this error :

```
Building in release mode
Compiling .pio\build\esp8266_d1minipro\src\DIYBMSServer.cpp.o
Compiling .pio\build\esp8266_d1minipro\src\PacketRequestGenerator.cpp.o
Compiling .pio\build\esp8266_d1minipro\src\SoftAP.cpp.o
Compiling .pio\build\esp8266_d1minipro\src\main.cpp.o
In file included from .pio\libdeps\esp8266_d1minipro\ESP Async WebServer\src/WebHandlerImpl.h:30:0,
                 from .pio\libdeps\esp8266_d1minipro\ESP Async WebServer\src/ESPAsyncWebServer.h:467,
                 from src\DIYBMSServer.cpp:26:
.pio\libdeps\esp8266_d1minipro\Time@src-012c38ce3e6dbfbcfc0f0e65ff7ff9e7/time.h:1:2: warning: #warning "Please include TimeLib.h, not Time.h.  Future versions will remove Time.h" [-Wcpp]
 #warning "Please include TimeLib.h, not Time.h.  Future versions will remove Time.h"
  ^
In file included from .pio\libdeps\esp8266_d1minipro\ESP Async WebServer\src/WebHandlerImpl.h:30:0,
                 from .pio\libdeps\esp8266_d1minipro\ESP Async WebServer\src/ESPAsyncWebServer.h:467,
                 from include/SoftAP.h:15,
                 from src\SoftAP.cpp:2:
.pio\libdeps\esp8266_d1minipro\Time@src-012c38ce3e6dbfbcfc0f0e65ff7ff9e7/time.h:1:2: warning: #warning "Please include TimeLib.h, not Time.h.  Future versions will remove Time.h" [-Wcpp]
 #warning "Please include TimeLib.h, not Time.h.  Future versions will remove Time.h"
  ^
Compiling .pio\build\esp8266_d1minipro\liba99\ESP8266WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp8266_d1minipro\liba99\ESP8266WiFi\WiFiServerSecureAxTLS.cpp.o
Compiling .pio\build\esp8266_d1minipro\liba99\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
Compiling .pio\build\esp8266_d1minipro\liba99\ESP8266WiFi\WiFiUdp.cpp.o
In file included from .pio\libdeps\esp8266_d1minipro\ESP Async WebServer\src/WebHandlerImpl.h:30:0,
                 from .pio\libdeps\esp8266_d1minipro\ESP Async WebServer\src/ESPAsyncWebServer.h:467,
                 from src\main.cpp:71:
.pio\libdeps\esp8266_d1minipro\Time@src-012c38ce3e6dbfbcfc0f0e65ff7ff9e7/time.h:1:2: warning: #warning "Please include TimeLib.h, not Time.h.  Future versions will remove Time.h" [-Wcpp]
 #warning "Please include TimeLib.h, not Time.h.  Future versions will remove Time.h"
  ^
In file included from src\PacketRequestGenerator.cpp:1:0:
include/PacketRequestGenerator.h:25:35: error: expected ')' before '*' token
   PacketRequestGenerator(cppQueue *requestQ) { _requestq = requestQ; }
                                   ^
include/PacketRequestGenerator.h:48:3: error: 'cppQueue' does not name a type
   cppQueue *_requestq;
   ^
src\PacketRequestGenerator.cpp: In member function 'void PacketRequestGenerator::pushPacketToQueue(PacketStruct*)':
src\PacketRequestGenerator.cpp:157:3: error: '_requestq' was not declared in this scope
   _requestq->push(_packetbuffer);
   ^
*** [.pio\build\esp8266_d1minipro\src\PacketRequestGenerator.cpp.o] Error 1
In file included from include/DIYBMSServer.h:24:0,
                 from src\DIYBMSServer.cpp:42:
include/PacketRequestGenerator.h:25:35: error: expected ')' before '*' token
   PacketRequestGenerator(cppQueue *requestQ) { _requestq = requestQ; }
                                   ^
include/PacketRequestGenerator.h:48:3: error: 'cppQueue' does not name a type
   cppQueue *_requestq;
   ^
In file included from include/DIYBMSServer.h:24:0,
                 from src\main.cpp:227:
include/PacketRequestGenerator.h:25:35: error: expected ')' before '*' token
   PacketRequestGenerator(cppQueue *requestQ) { _requestq = requestQ; }
                                   ^
include/PacketRequestGenerator.h:48:3: error: 'cppQueue' does not name a type
   cppQueue *_requestq;
   ^
src\main.cpp:232:1: error: 'cppQueue' does not name a type
 cppQueue requestQueue(sizeof(PacketStruct), 16, FIFO);
 ^
src\main.cpp:234:1: error: 'cppQueue' does not name a type
 cppQueue replyQueue(sizeof(PacketStruct), 8, FIFO);
 ^
src\main.cpp:236:54: error: 'requestQueue' was not declared in this scope
 PacketRequestGenerator prg = PacketRequestGenerator(&requestQueue);
                                                      ^
src\main.cpp: In function 'void serviceReplyQueue()':
src\main.cpp:406:11: error: 'replyQueue' was not declared in this scope
   while (!replyQueue.isEmpty())
           ^
src\main.cpp: In function 'void onPacketReceived()':
src\main.cpp:454:8: error: 'replyQueue' was not declared in this scope
   if (!replyQueue.push(&ps))
        ^
src\main.cpp: In function 'void timerTransmitCallback()':
src\main.cpp:475:7: error: 'requestQueue' was not declared in this scope
   if (requestQueue.isEmpty())
       ^
src\main.cpp:483:3: error: 'requestQueue' was not declared in this scope
   requestQueue.pop(&transmitBuffer);
   ^
src\main.cpp: In function 'void timerLazyCallback()':
src\main.cpp:1249:11: error: 'requestQueue' was not declared in this scope
       if (requestQueue.getRemainingCount() < 6)
           ^
*** [.pio\build\esp8266_d1minipro\src\main.cpp.o] Error 1
*** [.pio\build\esp8266_d1minipro\src\DIYBMSServer.cpp.o] Error 1
======================================================================================================== [FAILED] Took 5.15 seconds ========================================================================================================
Environment        Status    Duration
-----------------  --------  ------------
esp8266_d1minipro  FAILED    00:00:05.152
================================================================================================== 1 failed, 0 succeeded in 00:00:05.152 ================================================================================================== 
```

stuartpittaway, 2020-12-06:
Ok, I suspect you have an old version of the queue library. Take a look in the source code folder and delete the folder named .pio then try compile again.

donnib, 2020-12-07:
Yes that solved it, thx

closed 2020-12-07 by donnib