stuartpittaway/diyBMSv4Code #60: branch 256modules doesnt start up the softap properly but crashes

daromer2, 2020-12-07:
When using 256modules branch the softap doesnt work properly. It crashes the ESP after 2 seconds. I tried both the master branch right now that works fine and boots up and start debugging messages properly meanwhile 256 modules look like this:
[image]
As can bee seen it crashes and doesnt show the AP.  

Also why is the debugging ofsetted on the 256modules version compare to the master that looks nice?

daromer2, 2020-12-07:
Workaround is to install master branch and set config then update to 256 without erasing flash

donnib, 2020-12-07:
@daromer2 great workaround, this was the issue i was mentioning on facebook the other day. Now i am up and running :)

daromer2, 2020-12-07:
Should fix the issue but i didnt have time to see what differs or why it happens. Have alot to learn about the code first

stuartpittaway, 2020-12-08:
Change checked in

daromer2, 2020-12-08:
Works as it should now. Closing.

closed 2020-12-08 by daromer2