stuartpittaway/diyBMSv4Code #61: Version numbering on the slaves and masters

daromer2, 2020-12-07:
Would like to have something like : mi[i].BoardVersionNumber; But for the build version
So you easily can track if all nodes have the same revision uploaded or not.

Currently its so easy when testing that you forgot one and you wonder if you did upload the new firmware or not.

Or does this exist? I cant find it in the code of today send from the modules. I have only used built in ones in code but guess should be able to get some hash id of the uploaded data?

daromer2, 2020-12-07:
Perhapos this could be something: https://docs.platformio.org/en/latest/projectconf/section_env_build.html#dynamic-build-flags

If you're using Platformio, add this in platformio.ini file

```
build_flags = 
    !echo '-D GIT_VERSION=\\"'$(git rev-parse HEAD)'\\"'
And use it like this:

```
`Serial.printf("Firmware version: %s\n", GIT_VERSION);`

stuartpittaway, 2020-12-08:
Agreed, its quite difficult to get a number into the modules (16 bit needed for the packets t work), not everyone is downloading the code using GIT so any GIT commands also need to be optional.

I may have to revert to using Travis to do the builds and provide binary releases, I've avoided this so far as it was very time consuming so needs work to make automatic.

atanisoft, 2020-12-08:
@stuartpittaway instead of Travis take a look at GitHub Actions to do the binary builds.

daromer2, 2020-12-08:
I see this to be good in 2 cases
1. When coding/debugging so you know you got the correct running  versions everywhere
2. When having people using it and filing bug reports. Basically you can use a template to people to fill in

But i agree for the later it would be better to have a binary version number.

stuartpittaway, 2020-12-09:
Github actions seems easy to use, I've added in the code to use version numbers in both controller and modules.

Ready for test

atanisoft, 2020-12-09:
I've transitioned all of my projects to use GitHub Actions since it also allows a sanity test of PRs prior to merge. It is also a great way to bundle binaries for easy updating of devices via OTA.

stuartpittaway, 2020-12-09:
They are very useful, I'll also start to use them. I've got it to publish the artifacts as part of the build, so should make it easier to distribute code going forward.

I'll probably look to get those published to a "release" repo going forward.

atanisoft, 2020-12-09:
> I'll probably look to get those published to a "release" repo going forward.

I'd skip that and instead integrate the "release" tag process so you can tag a specific build as a release and the actions will build and upload the binaries to the published release.

stuartpittaway, 2020-12-21:
GITHUB actions added to automatically create releases on MASTER branch, GIT version is also embedeed in module and controller code.

closed 2020-12-21 by stuartpittaway