stuartpittaway/diyBMSv4Code #62: 256modules branch: I see regularly The controller is having difficulty communicating with the cell monitoring modules.

donnib, 2020-12-07:
I only have one module which i am testing.

3.32V
Range 0:
0mV
Packets sent:
103
Packets rec'd:
95
Roundtrip (ms):
352

I timed it to each 36-40sec this happens, some timer or watchdog ?

stuartpittaway, 2020-12-08:
Hi @donnib , the code is regularly changing in this branch, are you using the latest copy?

With only a single module, it is possible the controller is transmitting data faster than the module can accept it.  Do you have another module you can add to the string?

Are you seeing "OOS" errors?  (Out of Sequence)

donnib, 2020-12-08:
Yes i was running latest code yesterday. It's a good idea, i will flash one more module to test then get back. Where should i see the OSS errors ?

stuartpittaway, 2020-12-08:
The OSS errors appear next to the roundtrip/packet counters - if you don't see it thats a good thing :-)

The "controller having difficult" message occurs when no packets are received from the modules for 3x the roundtrip - about 1.5 seconds in your example.  The problem with this is that the controller only asks for the information every 8-10 seconds, so with only a single module, there is too much time between packets so the error appears.

Ideally this would automatically scale or be user selectable, but I didn't expect people to be using a single module on the BMS :-)

stuartpittaway, 2020-12-08:
Change submitted for testing

donnib, 2020-12-08:
@stuartpittaway of course the idea is not to run with just one module, so if that's the issue there is no issue. I thought maybe that was caused by something else so i wanted to help report that if that was the case ;)

daromer2, 2020-12-08:
I have the same issue with 14 modules i think. Always OOS error on the 5th module. Uploaded video. 
I checked debug on TX1 and i can see that the crc differs by one each time. 

The interesting part is that if i change out that module it seems that the error dissapears. 
I cant see any difference between them. Soldering is fine and i just repgroammed them as well to make sure running same version. Still same issue :)

stuartpittaway, 2020-12-08:
OOS really means that a reply was not received (skipped).  

I've just been helping a user on the forum with a similar issue.  Everything looked fine, but he remade a few of the JST connections/cables and the problem went away.

I think it may be time to look at different types of connectors on the modules as the JST sockets are a pain to work with.

daromer2, 2020-12-08:
Yeap this is used:  - framework-arduino-avr-attiny 1.4.1+sha.bf6f7dd
Yeah the cheap jst is a pity. The thing in my case is that i swapped out cables and modules where it happens. Its always on the same node. 
Gonna do another swap later to see

donnib, 2020-12-16:
@stuartpittaway I added another cell so i have two but i still get the same issue.

stuartpittaway, 2020-12-16:
Hi @donnib as you can probably see from the branches the code has under gone rapid development over the past 4 weeks.

Can you reprogram the controller and modules to use the code from this branch.  
https://github.com/stuartpittaway/diyBMSv4Code/tree/modulepwmchanges

Note that the "upload file system image" is NOT LONGER required for this branch!

donnib, 2020-12-16:
Yes I can see and I just merged the two branches myself and build them together. I can try the branch you suggest. So which specific flash command should I use in this branch and why is it different?

stuartpittaway, 2020-12-16:
I've completely changes the way the modules balance, which is why we can now energy count how much has been burnt off.

Just program the code to the boards as you have been doing for previous modules and controllers.

donnib, 2020-12-16:
@stuartpittaway everything seems good now. no more errors with two modules and the latest changes in the https://github.com/stuartpittaway/diyBMSv4Code/tree/modulepwmchanges branch. I close this issue.

closed 2020-12-16 by donnib