stuartpittaway/diyBMSv4Code #63: Can't compile attiny841_V421 firmware

Hartmut01, 2020-12-13:
I'm trying to compile the Firmware for the Battery modules, but platformio is not able to make the files.
When I try to compile the Attiny841_V421 which I think i need, because I got my PCBs made from JLCpcb, I get the below error messages.
```
> Executing task in folder ATTINYCellModule: C:\Users\Hartmut\.platformio\penv\Scripts\pio.exe run --environment attiny841_V421 <
Processing attiny841_V421 (platform: atmelavr; board: attiny841; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/attiny841.html
PLATFORM: Atmel AVR (1.8.1) > Generic ATTiny841
HARDWARE: ATTINY841 8MHz, 512B RAM, 8KB Flash
PACKAGES:
 - framework-arduinoavr 1.10620.2 (1.6.20)
 - toolchain-atmelavr 1.40902.1 (4.9.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PacketSerial> 1.4.0+sha.840272c
|-- <FastPID> 1.3.1+sha.1792d23
|-- <settings>
|   |-- <crc16>
|   |-- <EEPROM> 2.0
|-- <crc16>
|-- <Steinhart>
Building in release mode
Compiling .pio\build\attiny841_V421\src\main.cpp.o
Archiving .pio\build\attiny841_V421\lib793\libFastPID.a
Archiving .pio\build\attiny841_V421\lib0ea\libcrc16.a
Archiving .pio\build\attiny841_V421\liba67\libsettings.a
Archiving .pio\build\attiny841_V421\lib3f6\libSteinhart.a
Compiling .pio\build\attiny841_V421\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\attiny841_V421\FrameworkArduino\Print.cpp.o
Compiling .pio\build\attiny841_V421\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\attiny841_V421\FrameworkArduino\Tone.cpp.o
In file included from src\main.cpp:40:0:
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h: In member function 'void PacketSerial_<EncoderType, PacketMarker, ReceiveBufferSize>::begin(long unsigned int)':
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h:88:16: error: no match for 'operator!' (operand type is 'HardwareSerial')
         while (!Serial) {;}
                ^
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h:88:16: note: candidate is:
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h:88:16: note: operator!(bool) <built-in>
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h:88:16: note:   no known conversion for argument 1 from 'HardwareSerial' to 'bool'
Compiling .pio\build\attiny841_V421\FrameworkArduino\WInterrupts.c.o
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h: In member function 'void PacketSerial_<EncoderType, PacketMarker, ReceiveBufferSize>::begin(long unsigned int, size_t)':
Compiling .pio\build\attiny841_V421\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\attiny841_V421\FrameworkArduino\WString.cpp.o
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h:105:24: error: no match for 'operator!' (operand type is 'HardwareSerial')
                 while (!Serial1) {;}
                        ^
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h:105:24: note: candidate is:
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h:105:24: note: operator!(bool) <built-in>
.pio\libdeps\attiny841_V421\PacketSerial\src/PacketSerial.h:105:24: note:   no known conversion for argument 1 from 'HardwareSerial' to 'bool'
In file included from lib\settings/settings.h:25:0,
                 from src\main.cpp:51:
C:\Users\Hartmut\.platformio\packages\framework-arduinoavr\libraries\__cores__\tinymodern\EEPROM/EEPROM.h: At global scope:
C:\Users\Hartmut\.platformio\packages\framework-arduinoavr\libraries\__cores__\tinymodern\EEPROM/EEPROM.h:145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
 static EEPROMClass EEPROM;
                    ^
*** [.pio\build\attiny841_V421\src\main.cpp.o] Error 1
================================================== [FAILED] Took 1.37 seconds ==================================================

Environment     Status    Duration
--------------  --------  ------------
attiny841_V421  FAILED    00:00:01.370
============================================= 1 failed, 0 succeeded in 00:00:01.370 =============================================The terminal process "C:\Users\Hartmut\.platformio\penv\Scripts\pio.exe 'run', '--environment', 'attiny841_V421'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Also Visual Studio shows me two problems:
`no match for 'operator!' (operand type is 'HardwareSerial')[88,16]`
`no match for 'operator!' (operand type is 'HardwareSerial')[105,24]`

Can you help me figure out what I did wrong?

Thanks in advance for all the energy you put in this project :)

Hartmut01, 2020-12-13:
Hello,

Yes I did as you said in your Video clone with the git Clone Command the master branch.
I only installed Platformio extension nothing else for Visual Studio Code.
Arduino 1.8.4 is also installed.
building the Code for the ESP works perfectly fine.

I just uninstalled Visual Studio Code completely with alle the folders tha are not uninstalled during nochmal uninstall.
Installed the newes Visual Studio Code again. and the Platformio extensoin and the same error as before.

Do I need the Arduino extension aswell? when I install this extension aswell I always get an error that there is a conflict linking to this page: https://github.com/platformio/platformio-vscode-ide/issues/118
And still the Firmware does not compile.

If I look in my Workspace Folder in the  Path: `diyBMSv4Code\ATTINYCellModule\.pio\build\attiny841_V421`
the is a Folder called: Frameworkarduino

Thanks for all your help :)

stuartpittaway, 2020-12-13:
Have you followed the instructions on that issue page and disabled the vscode extension?

Hartmut01, 2020-12-13:
If I understand the instruction correctly I should disable/Uninstall the Arduino extension (which I did) 
and install the C/C++ Extension Pack(1.0.0).
still the same error as before.

Hartmut01, 2020-12-13:
I finally got i working.
I found through this Video: https://www.youtube.com/watch?v=8cTo3iLBBcY
That ther was a Time where the Arduino IDE did not understand the Command `!Serial`

I had Arduino IDE version 1.8.6 Installed which also did not understand this command.

Solution:
Install newest Arduino IDE 1.8.13
Uninstall VS Code
Delete the Folders:
```
"%UserProfile%\.platformio"
"%UserProfile%\.vscode"
"%AppData%\Code"
"%AppData%\Visual Studio Code"
```
Reboot

Install VS Code new
Install platformio extension

Try again.

And now I can Compile again 👍 
Thank you very much for your support and hinting me in the right direction :)

closed 2020-12-13 by Hartmut01