stuartpittaway/diyBMSv4Code #68: Platform IO drives me nuts

donnib, 2020-12-16:
I am trying to try some of the changes you @stuartpittaway made. I am trying to put the new updates on the controller but no matter what i did it was the same version so i did a Erase Flash and now no matter which branch i Build Filesystem image then Upload Filesystem image nothing happens, i mean everything runs fine and it flashes the ESP but the AP never comes up. I tried master and i tried the other branches. None fail but the AP does not come up. Are these bugs in Platform IO ?

stuartpittaway, 2020-12-16:
Hi donnib, if you just flash the file system image and the program, the ESP will remember the wifi settings so the AP never reappears, as its already connected to your wifi !

I'm currently working on using GITHUB ACTIONS to automatically compile the code and provide you with a complete ready to program file, so you won't need to venture near platformio, a lot of people hate it, personally, its 1000x better than the Arduino IDE !

donnib, 2020-12-16:
Yes I know but the controller doesn't connect to my own WiFi so either. You say flash and program, is that a two step ? I know that platform io is much better than Arduino ide. Regarding automatic build system that sounds good and if it also works on the branches even better.

stuartpittaway, 2020-12-16:
In platform Io you need to click upload and the upload file system image. Two different steps.

The auto building is already working for the branches, feel free to take a look in GitHub actions.

closed 2020-12-16 by stuartpittaway

donnib, 2020-12-16:
@stuartpittaway should i not see artifacts somewhere e.g a binary file i can use to flash with ?