stuartpittaway/diyBMSv4Code #70: Wrong voltage reading

axwell, 2020-12-24:
Hello, my packet from jlcpcb just arrived, everything works but cell voltage reading is 1.3 volts, but the cell has 4.1 V
Calibration multiplier does not allow values over 5.

What could be the issue for this wrong voltage reading ?

Thanks.

stuartpittaway, 2020-12-24:
Hi, what version of the board is it?  Was it assembled by JLCPCB?  Did you replace any components with another type?  (TL432?)

From the description is sounds like a bad solder joint or short  The calibration value should be around 2.2

axwell, 2020-12-24:
V4.21 
Hmm if you are referring to D1, i was not available at jlcpcb so i manually soldered this:
https://www.tme.eu/ro/details/lm4040bim3-2.5/surse-tensiune-de-referinta-circuite/texas-instruments/lm4040bim3-2-5-nopb/

I just saw now that this has 2.5 voltage reference instead of 2, i can't find 2.0 anywhere for Romania.
Is it possible to do the changes in the code ?

stuartpittaway, 2020-12-24:
Version 4.21 of the board doesn't use LM4040 part!

Version 4.21 uses a TL432 1.25V reference not the LM4040.  

I think you have mixed up the B.O.M for the V4 and the V4.21 boards - they are different.

axwell, 2020-12-24:
The components on the board are ok, i just took D1 from master branch.
Unfortunately i can't find TL432G-A-AE3-R in romania :( , can i use LM4040 with some software changes ? or do you recommend a drop in replacement ?

Thank you !

stuartpittaway, 2020-12-24:
The lm4040 won't work, there are other variants of the tl432 which might. Can you get any similar parts?

axwell, 2020-12-24:
yes, if i can find it on https://www.tme.eu
Can you recommend equivalent part ?

stuartpittaway, 2020-12-24:
You will have to check that the pins are the same as the original part, but this looks like it will do the job.

AZ431LBNTR-G1

axwell, 2020-12-24:
Pins are ok, i ordered the components and i will be back with details.
Thanks ! Happy holidays :)

20after4, 2021-01-01:
I'm also getting very low readings. I had to increase the calibration value to 6.8 on all 5 of the v4.21 modules that I ordered from JLCPCB. (Note: to increase the calibration value you just have to override the Max=5 in the html form)

stuartpittaway, 2021-01-01:
> I'm also getting very low readings. I had to increase the calibration value to 6.8 on all 5 of the v4.21 modules that I ordered from JLCPCB. (Note: to increase the calibration value you just have to override the Max=5 in the html form)

Thats definately a problem, the calibration is normally around 2.2, whats changed on the PCB you ordered? did you have to substitute some parts?  Is the voltage reference the correct part?

20after4, 2021-01-01:
The component at d1 is entirely missing. I guess that’s the issue?

stuartpittaway, 2021-01-01:
Yes, I'm surprised its working at all - the voltage measurements are not going to be accurate, so don't use it in a real power wall/battery bank until D1 is installed !

20after4, 2021-01-02:
So d1 just needs to be an accurate 1.25v reference? Is there a reason to
choose 1.25 over 2.0v for reference voltage?

axwell, 2021-01-03:
AZ431LBNTR-G1 as replacement part works.

stuartpittaway, 2021-01-21:
Thank you for confirming issue and replacement part.

closed 2021-01-21 by stuartpittaway