stuartpittaway/diyBMSv4Code #72: Check of inverter voltage / frequency as rules

DGrebe, 2020-12-27:
Hi,

in my setup I have a double relay that is switching load between grid and inverter in order to have a save power supply in case of low battery or inverter fails.

This relay should be controlled by the ESPController rules.

As an additional measure, I would like to measure the output of the inverter (esp. voltage and frequency).
These measures should be evaluated in the rules section.

I see two options to implement this via the ESPController:
1. Direct modbus reading of SDM energy meters (https://github.com/reaper7/SDM_Energy_Meter) + keep everything required directly on the controller - would require a SW Serial connection and HW addition of the Controller board to include the modbus interface
2. Use MQTT subscribe to receive the data from the energy meter + flexible and decoupling features, easier to implement, no hardware changes, - could distort the module communication, depends on external communication

Alternatively, one could use a separate industrial voltage control circuit (e.g. Eltako 2553803 ELTA NR12-001) + more fail save - loose flexibility in rules, no impact of the battery parameters on the switching.

Any interest in or suggestions for the implementation. 
Any suggestions or hints how others implemented security features controlling the inverter / power output? (off-topic)

Regards,

Dirk

stuartpittaway, 2020-12-27:
Hi Dirk, the new controller uses an ESP32 and that allows the extension of whats possible, including RS485 and CANBUS for direct reading from inverters/chargers.  This is currently work in progress.

There is also a separate voltage and current shunt design which will integrate with the new controller.

DGrebe, 2020-12-27:
Hi Stuart,

Thanks, I will look into this.

Regards

Dirk

DGrebe, 2020-12-27:
Hi Stuart,
I have found the code, but no information on the ESP32 controller. Do you plan a new controllerboard or does it fit into the existing one.

stuartpittaway, 2020-12-27:
The controller is all new. Take a look at the openenergymonitor forum for picture and information. It's not released the the public just yet, still testing.