stuartpittaway/diyBMSv4Code #74: Wemos di mini clone

Michaelillingby, 2020-12-31:
I have all 55 of my attiny boards and 5 controllers sitting on my desk. great, next step programming. I have 5 di mini clones the say WAVGAT on them, and I was hoping to use them since Adam Welches video says he could do it changing a few flags, My code is different than his and I am stuck as to how to make this work with my boards. can you offer some support/advice please?
Thanks in advance
[image]
[image]
[image]

I really hope I can use this version, as it will take me weeks to get new ones in my area, thank you again

stuartpittaway, 2020-12-31:
Have you tried programming the ESP8266 code to this board as per the YouTube videos?

It should just work - the Pro version isn't technically needed any more.

The PIN out on the boards photoed is the same as the "pro" version, so I suspect will work.

Michaelillingby, 2020-12-31:
yeah, the videos are like 6 months old and the software doesn't quite look the same as what I downloaded a day ago. in the code, if i'm reading it correctly, the is chip support for the D1 pro mini, and the 32. what do I need to change? are you referring to a different video perhaps? can you provide a direct link please?

stuartpittaway, 2020-12-31:
No changes needed, just plug it into a USB port and program.  You may need to remove/change the COM port setting in platformio.ini file, but thats it.

Michaelillingby, 2020-12-31:
okay will try again tonight

Michaelillingby, 2021-01-04:
same problem
here is the output

> Executing task in folder ESPController: C:\Users\foxan\.platformio\penv\Scripts\pio.exe run --target upload --environment esp8266_d1minipro <

Processing esp8266_d1minipro (platform: espressif8266@2.6.2; board: d1_mini_pro; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_pro.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 mini Pro
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - tool-mklittlefs 1.203.200522 (2.3)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/bakercp/PacketSerial.git
FileNotFoundError: [WinError 2] The system cannot find the file specified:
  File "C:\Users\foxan\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\foxan\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\foxan\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\foxan\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\foxan\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\foxan\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\foxan\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\foxan\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\foxan\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\foxan\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\foxan\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\foxan\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\foxan\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
    force=force,
  File "c:\users\foxan\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\foxan\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\foxan\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55:
    src_dir, remote_url, tag, silent
  File "c:\users\foxan\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137:
    self.configure()
  File "c:\users\foxan\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146:
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\foxan\.platformio\penv\lib\site-packages\platformio\proc.py", line 113:
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\foxan\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\foxan\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
============================================== [FAILED] Took 12.92 seconds ==============================================

Environment        Status    Duration
-----------------  --------  ------------
esp8266_d1minipro  FAILED    00:00:12.917
========================================= 1 failed, 0 succeeded in 00:00:12.917 =========================================
The terminal process "C:\Users\foxan\.platformio\penv\Scripts\pio.exe 'run', '--target', 'upload', '--environment', 'esp8266_d1minipro'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Michaelillingby, 2021-01-04:
now to confirm that is is being detected, I uploaded BLINK using arduino IDE, that works no problem. I tried the wifi scan example and that works too.

I am at a loss, what can I do. I should mention that this is all newly installed software, so if you didnt tell me to install it, I didn't. if I wasn't prompted to install it, I haven't.

I need this BMS asap, I have been without one for far too long, thanks

Michaelillingby, 2021-01-04:
this id what PlatformIO detects as my hardware
It is NOT a PRO from what I can tell, I believe it is just a D1 Mini  -  4MB 80-160MHz

Port | Description | Hardware
-- | -- | --
COM3 | USB-SERIAL CH340 (COM3) | USB VID:PID=1A86:7523 SER= LOCATION=1-1.1

Michaelillingby, 2021-01-04:
oh yeah, I also didn't have to change the com port as it was already set to COM3

atanisoft, 2021-01-04:
@Michaelillingby do you have git for windows installed?

Michaelillingby, 2021-01-04:
not sure what that is

atanisoft, 2021-01-04:
@Michaelillingby then it is likely a safe bet that you don't have it installed. I'd suggest install git for windows: https://git-scm.com/download/win. You can likely use the default settings in the installer.

stuartpittaway, 2021-01-04:
Hi @Michaelillingby the errors reported are platformio issues.

I think if you install GITHUB DESKTOP - https://desktop.github.com/

that will resolve the issue you have.

atanisoft, 2021-01-04:
@stuartpittaway I'd suggest log an issue against PIO not installing it's required dependencies. Maybe they will fix it..

Michaelillingby, 2021-01-04:
is there a way to upload with the arduino IDE?

Michaelillingby, 2021-01-05:
Ok, I solved it!
The problem was the upload speed was a billion times too high for some reason. I changed it to 115200 and it took about 45 seconds to upload and now it works.
I did this during my free time at work and connected it to my work wifi and it presented itself nicely. now new challange, how do I change the wifi it connects to? Is there a hard reset I can do to get back the the "softap" I think it's called? the wifi setup page stand alone server?

stuartpittaway, 2021-01-20:
> Is there a hard reset I can do to get back the the "softap" I think it's called? the wifi setup page stand alone server?

Yes, RESET the ESP module and when the LED lights up connect the "WIFI RESET" pins together  - you only have a few seconds.

This will wipe the WIFI settings and force you to select again.

closed 2021-01-20 by stuartpittaway