stuartpittaway/diyBMSv4Code #78: Not sure if this is the right area for this.

Michaelillingby, 2021-01-11:
I posted this on the open energy forum, not sure if it was supposed to go there or here. advise and I'll delete the wrong one.

Need a little help here with a few things;
1) I run an odd sized bank. I have 17S2P that I wish to monitor. How do I best do this? I currently have 17 modules hooked up and running, 0/0-1/0 and thats working just fine, sorta. How do I add my other 17S bank? Do I need to run a second loop with its own controller and IP address? Is there a possibility of changing the SERIAL BANK SIZE? like a drop down, or a text field
2) How can I can the voltage RANGE on the main screen? I run LiFePO4 and my MAX is 3.65vdc, so I would like the max voltage say 3.75vdc. The reason is that screen is not well used all being at the bottom.  Is there some code that I can edit, or is this hard coded? and option on the settings screen would be nice.
3) I am getting some errors, and I am not sure why, it works mostly, hangs up sometimes. I can Leave a window open to the settings page in the background so I have access to the reset button, and that works most times. When I say hang up, I mean that the busy circle in the middle just keeps rotating and never fully loads. 
on my modules page I see many bad packets on some but not all modules heres a clip;
Bad packet
count

Bank	 Module 	Voltage 	V. Min 	V. Max 	Temp Int °C 	Temp Ext °C 	Bypass 	PWM	 Bad packet count
0	0	3.321	3.316	3.332	13	0	1
0	1	3.344	3.339	3.350	13	0	3 
0	2	3.318	3.318	3.323	13	0	0
0	3	3.321	3.316	3.326	12	0	8
0	4	3.314	3.309	3.324	12	0	12
0	5	3.319	3.314	3.325	12	0	0
0	6	3.310	3.305	3.315	12	0	3
0	7	3.315	3.310	3.320	12	0	2
0	8	3.325	3.314	3.325	11	0	1
0	9	3.322	3.312	3.322	11	0	13
0	10	3.314	3.309	3.319	12	0	116
0	11	3.313	3.308	3.318	12	0	86
0	12	3.314	3.314	3.319	12	0	25
0	13	3.316	3.310	3.321	12	0	46
0	14	3.313	3.313	3.319	12	0	49
0	15	3.310	3.310	3.321	12	0	50
1	0	3.329	3.329	3.344	12	0	57

Notice that the mass of errors around cell 0/9. What is the likely cause of this?

4) Also, could you add a rule that uses the temp Int? This would be nice in addition to cell temp. Ie turn on a fan for high temp, or disable charging for cold cells etc.

5) Do the wires between the modules need to be twisted? If so how many twists per inch/cm?

6) Self powered controller? It  there a possibility of including a small voltage reg to buck the pack voltage down to 5v? The other advantage is a controller based voltage sense.

One last thing, when I ordered my boards assembled, there were two parts they didn’t have, one was the attiny, and the other was the voltage ref I believe. I got a replacement ref from you when I ordered; Was i supposed to change any code when I swapped this v ref chip?

Thats all I have for now.

PS Stuart, I have exposed my esp via a port forward so you can poke around if need be, PM me for the link so you can see the errors I am getting. Due to no security on the webpage I won’t post this publicly.

Thanks in advance, Michael.

stuartpittaway, 2021-01-11:
Hello Michael,

Good news for you, the next release of the diyBMS code contains most of the items you want resolving!

If you want to try it now, the code is here, you program it to the controller and modules just as you did when you first installed it.  You will need to reprogram all the module ATTINY chips as well as this is a major change to the way things work under the covers.
https://github.com/stuartpittaway/diyBMSv4Code/tree/modulepwmchanges

Hopefully I can answer the questions...
**I run an odd sized bank. I have 17S2P that I wish to monitor. How do I best do this?** 

Use the new code, it supports up to 100 modules, in a combination of up to 32S and 16 parallel banks.  So 17S x2 strings (banks) will work fine.

**I am getting some errors, and I am not sure why, it works mostly, hangs up sometimes.** 

The ESP chip can get bogged down with lots of web requests.  The new code doesn't need the "upload filesystem image" option, which means that the new website is much quicker at loading.  This should get rid of the problem you are seeing.

**on my modules page I see many bad packets on some but not all modules heres a clip;**

I don't know how many packets have gone through the system to indicate if this is bad or not,  would expect a few bad packets every now and then, especially when balancing.  Once a module generates a bad packet it will move through the others and they will record the error as well, so it looks worse than it really is.  

**Notice that the mass of errors around cell 0/9. What is the likely cause of this?**

I would be tempted to check the module with the highest count.  Possibly a bad solder joint or lose cable/JST connector.

**How can I can the voltage RANGE on the main screen?**
Yes - its in the next version!

**Also, could you add a rule that uses the temp Int?**
Yes - its in the next version!

**Do the wires between the modules need to be twisted? If so how many twists per inch/cm?**
Yes - I've never actually measured the twists, they don't need to be super tight, I'd probably say 2 or 3 twists per inch.

**Self powered controller?** 
This is quite difficult to do as everyone has different voltage ranges on their cells.  The easiest way to do this is with an external step down power supply/DC to DC converter.

What voltage reference chip did you end up using to replace the TL432?  If it has a similar voltage (about 1.24V), then nothing needs to be changed.

Michaelillingby, 2021-01-11:
> on my modules page I see many bad packets on some but not all modules heres a clip;

I don't know how many packets have gone through the system to indicate if this is bad or not, would expect a few bad packets every now and then, especially when balancing. Once a module generates a bad packet it will move through the others and they will record the error as well, so it looks worse than it really is.

It was after around 23,000 packets sent, 18,000 received. does that loss seem excessive?

https://jlcpcb.com/parts/componentSearch?searchTxt=C34674 this is the part I swapped in for the voltage ref.

Last question, for now... I did see that someone created a board for use specifically for the Leaf cell.. I was wondering if you would be willing to design a board for use with the 280aH Prismatic cells lots of us are using these days? I could supply a cad for the terminals if that helps. Seems that the terminals for the different sizes are all the same with only the height and thickness changes. It would make things so much cleaner to install if I could just place it on top of my busbars, also no voltage drop across the leads.
I am completely ok with just buying neew modules that bolted right on, as the fact that I never installed the 6 pin header on means reprogramming them is going to suck. I would rather toss them out, or give them to someone else that wants them.

Last thing, is there a way that I can email you privately to get you a link to my module? (I don't want to post it publicly since there is no username/password for the module. I would really like to have you look at the numbers and see if you can figure out these errors I am getting. You can email me at michael.i@me.com thanks

Michaelillingby, 2021-01-11:
it's too bad that none of these board makers can also supply the attiny so we didn't have to solder that.
Also, if making that custom board is something you would be ok with, I have some other design ideas that I would like to suggest. hit me up.

stuartpittaway, 2021-01-11:
Did you see the round board design I recently did for the LTO cells?

Michaelillingby, 2021-01-11:
no, i did not. where do you post these marvelous creations of yours?

stuartpittaway, 2021-01-11:
It's on the forum...

Michaelillingby, 2021-01-11:
hrm, I must've skipped it

Michaelillingby, 2021-01-12:
ok, I saw it and created a new topic for the LiFePO4 version. 😉 Love thiscommunityy

stuartpittaway, 2021-01-12:
"It was after around 23,000 packets sent, 18,000 received. does that loss seem excessive?"

Yes I think this is excessive.  When you see packets being lost, look and see if the green led's are flowing along the cells correctly - if they get to the end, then its likely a problem with the last connection to the controller.  Also try twisted cables between the modules.

I won't get time to look at the setup you have on a 1:1 basis, so I'd recommending closing that port forward as that isn't a good idea with a BMS - anyone on the internet could drain the cells!

stuartpittaway, 2021-01-27:
Hi @Michaelillingby have you tried the recently released code on your setup?

Michaelillingby, 2021-01-27:
no i have not tried the new code yet, as I do not have an easy way to reprogram the cell modules. You see when I built the boards, I didn’t consider that I may need to flash them more than once and never added the connectors. this saved the labor and cost of installing them, but makes future uploading a pain in the ass. I do plan on doing your NEW controller once released and will fix the boards then for sure.

stuartpittaway, 2021-01-27:
I don't add the ISP header pins either - I use a pogo pin adapter and just push it/hold it against the contacts.

I built my own, but this is similar - https://www.sparkfun.com/products/11591

https://forum.arduino.cc/index.php?topic=492856.0

Michaelillingby, 2021-01-27:
wow, GMTA I just bought a bag of 100 pogo pins. need to make a JIG to line everything up. I’ll let you know once I get my sh*t together over here.

I have noticed that I don’t need to reset the wemos much at all if I ensure to only log into it from one computer at a time. If I have it logged in from more than one machine, it seems to crash.

stuartpittaway, 2021-01-27:
> If I have it logged in from more than one machine, it seems to crash.

May be worth raising another ticket/issue on that, it shouldn't crash, but the ESP8266 is very low powered for what its actually doing.

stuartpittaway, 2021-01-27:
> need to make a JIG to line everything up.

If you have an empty module PCB board (or a spare you can remove the parts from) then that makes a perfect jig.