stuartpittaway/diyBMSv4Code #79: Negative temperatures

Daemon256, 2021-01-12:
The balancer stops working at negative temperatures. If you heat the thermister with your finger, work is restored.

stuartpittaway, 2021-01-12:
In the picture you had a temperature of "-8", was this actually the correct temperature at the module?

Daemon256, 2021-01-12:
Yes, the temperature is correct, it's -15 outside now, the modules are in the shed, so it's a little warmer.

stuartpittaway, 2021-01-12:
Hi @Daemon256 , looks like a classic casting variable from int to unsigned int issue.

I've updated the code, if you can test that would be very helpful.

Daemon256, 2021-01-12:
Now you have to reflash all modules? It will take time.

stuartpittaway, 2021-01-12:
Just do 1 for now and ensure it starts working again.

Daemon256, 2021-01-12:
I can check tomorrow

Daemon256, 2021-01-13:
Confirmed. Works.

closed 2021-01-13 by Daemon256