stuartpittaway/diyBMSv4Code #80: DIYBMS newcontrolleresp32 does not start anymore

DGrebe, 2021-01-16:
Hi Stuart,

I try to continue with the modbus MQTT & influx publishing.
Somehow I can't start the controller anymore:

----
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400949fb
file: "src/HAL_ESP32.cpp" line 54
func: esp_err_t HAL_ESP32::writeByte(i2c_port_t, uint8_t, uint8_t, uint8_t)
expression: i2c_master_cmd_begin(i2c_num, cmd, pdMS_TO_TICKS(100))
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400949fb
file: "src/HAL_ESP32.cpp" line 107
func: void HAL_ESP32::Led(uint8_t)
expression: writeByte(I2C_NUM_0, TCA9534APWR_ADDRESS, TCA9534APWR_OUTPUT, TCA9534APWR_Value)
----

I saw the changes (e.g. RS485 port) and adapted my breadboard setup.
I have attached an PCF8574A via I2C_NUM_0 (Pins 22/23) and address 0x38 and got it working with a basic blink sketch. According to my information, TCA9534A and PCF8574A are compatible.

What should I do
- disable the I2C logic?
- can I support in any other form? rewriting the modbus communication to get rid of the SDM lib?
- do nothing and let you work?

Cheers,
Dirk

stuartpittaway, 2021-01-16:
Hello, that particular branch is for the new controller hardware and the code has significantly changed compared to the ESP8266 version of the controller - there were  a lot of conditional #defines but these were adding a lot of complexity to the code, you can see I've made a lot of changes over the past week or so.

Yesterday I added in code to support external SD card logging.

I'm currently having a think about how to do the RS485/MODBUS and the general input/output/rules because it feels that a more modular approach is needed to allow values from RS485 or CANBUS (or MQTT) to be used as inputs to the rules and also as outputs.

I think some sort of configuration file (probably JSON) is needed to support a vast number of different RS485/CANBUS devices so the end user can select from a list of supported devices and doesn't need to worry about configuration - this also removes the need to have the entire user interface catered for manually entering names and port numbers of the devices.

Storing this configuration in the FLASH memory is also troublesome - it would be much better in a file that can be downloaded/backed up as there are now a lot of settings.

I've not published the circuit or PCB yet for the new controller as its still in testing, however here is a sneak peak of the components I'm using and how the circuit looks.

[circuit.pdf](https://github.com/stuartpittaway/diyBMSv4Code/files/5824276/circuit.pdf)

* TCA6408AQPWRQ1 is used to drive relays and provide 4 input or output pins
* TCA9534A is used for on board drivers for LED's and emergency stop input
* The TX2 and RX2 and 6N137S1(TA) are likely to be removed

Everything else has worked out well, and I'm just about to order some more prototype PCBs to fix a few issues I've found

DGrebe, 2021-01-16:
Hi Stuart,

Thanks for the quick feedback and information on the board.

I also though about putting the Modbus Config into a JSON Config file, that can be loaded via the UI, but the config should be stored on the Controller to be available after reboot.

If OK, I would work on this.

Cheers,
Dirk

stuartpittaway, 2021-01-16:
If you can put the JSON configuration files into the "data" directory, then it will get compiled up into the LittleFS filesystem and can be uploaded into the ESP32 using the "upload file system image" option.

DGrebe, 2021-01-16:
I have commented out the I2C calls in my code to get my setup running and enabled the Modbus reading again.

closed 2021-01-16 by DGrebe