stuartpittaway/diyBMSv4Code #84: Flash using pio + "arduino as ISP"

corbolais, 2021-01-20:
Hi,
Just flashed the `ATTiny841` hex using `pio` and an "Arduino as ISP", here I used an UNO. For future reference in case anyone wants to quickly repeat this process I'm opening this issue. Maybe one could integrate it into the documentation?

**Firstly**, flash the bootloader (prefererably the "new" variant) to your *duino to be able to use it as ISP programmer. This can be done using the Arduino IDE (didn't get to flash it using `pio run -t bootloader`):
  1) Connect USB cable to the Arduino UNO and nothing else is connected at this point
  2) Tools->"Programmer: Arduino as ISP"
  3) Tools->"Flash bootloader" command.Sometimes this didn't work here, so an alternative is to select the "arduino ISP" example sketch compile and upload

**Secondly**, connect the Arduino ISP to your ATTiny board like this:
```
Arduino ISCP pin/
      header pin -> ATTiny board ICSP (pinout per schematic)
  ----------------------------
    1 MOSI / D11 -> 1 MOSI
    4 MISO / D12 -> 4 MISO
    3 SCK  / D13 -> 3 SCK
    2        VCC -> 2 VCC
    6        GND -> 6 GND
             D10 ->   /RST <== This is crucial
    5        RST -> 5 NC (!)
```

This may be done using the Arduino ISCP header or directly using the pins as shown in parenthesis. [Arduino ICSP pinout](https://www.arduino.cc/wiki/static/969690fbdaf4ea96844be9a21435f781/5a8a8/ICSPHeader.jpg).
Usually, you also need a 10uF cap across D10 (Arduino RST) and GND.
(For programming, in this case, CS (pin 6:PA7 or pin 11:PA2) on the ATTiny is not required to be pulled down.)

Thirdly, flash the firmware either using `avrdude` directly:
`avrdude -C ~/./avrdude.conf -p attiny841 -P /dev/ttyUSB0 -b 19200 -c stk500v1 -U flash:w:.pio/build/attiny841_V400/module_firmware_attiny841_V400_400.hex:i`
Just replace the hex filename with the one *you* need and the correct `ttyUSB?` device.

To flash using `pio run -e attiny841_V400 -t upload --upload-port /dev/ttyUSB1 -vv` one would need to modify `platformio.ini`:

HTH

LeighAS, 2023-03-06:
I may have to try this, i have failed a few times trying with usbasp devices

EDIT my new usbasp device arrived. same problem as before observed. tried -F upload flag, tried -B16 upload flag. all to no avail. Any advices here Stuart. Last thing to try might be the tinyisp like Adam Welch does but honestly think that the slaves may be broken too.

I struggled last night to flash the 4.21 slaves
Uploading .pio\build\V421\module_fw_V421_attiny841_421_eF4_hD6_l62.hex

avrdude: error: program enable: target doesn’t answer. 1
avrdude: initialization failed, rc=-1
Double check connections and try again, or use -F to override
this check.

I have a usbasp device which is showing as good in device manager and is being recognised in platformio in vscode, I have triple checked the six wires that go from the icsp headers and when i attempt upload it give the message shown above. any advices would be appreciated. thanks. I get one green flash on the 4.21 slave board when the upload is attempted.

I have now put an order on amazon for the usbasp linked in the description of this video. hopefully there is just some difference in my old usbasp device that i cant see, but i am doubtful.

LeighAS, 2023-03-07:
> Are you using the older (small) controller PCB? The newer one includes a
> TINY841 programmer through the web interface.

Yes its the older one about 2 years ago the v4.21 and usbasp programmer
through the 6 pins on the slave.
Havent tried a tinyisp flasher like adam welch yet but that my last
possible thought to get these working.

LeighAS, 2023-03-07:
[image]

LeighAS, 2025-06-03:
A bit more progress with Mqtt I've realised v4 never had auto discovery support. So until I get an esp32 I can have that. But I have manually configured my sensor in a yaml file now and have most of what I can see from the # topic listener.

[image]

So I'm getting pretty close to the done on it. Going to flash another 8 modules and make it a 16s pack 1p instead of 8s2p and I will have another go at my pogo pin flasher idea. Anyway thanks for all the hard work here.

stuartpittaway, 2025-06-03:
It looks like you've done most of the hardwork!!