stuartpittaway/diyBMSv4Code #89: Warning: Modules are configured with different bypass temperatures

Jani4ik, 2021-01-22:
Updated but the latest firmware (Release-2021-01-21-14-48) I get a message: “Warning: Modules are configured with different bypass temperatures”.
All modules are set to the same temperature 45 degrees Celsius.
Video - https://youtu.be/GBZ7nKIU9KU

stuartpittaway, 2021-01-23:
Thank you for the video, I've built a test using 4 modules and 1 controller, but I can't get the warning message to appear.

Can I ask you to reprogram the ESP8266 using the latest code/release? - there isn't any need to reprogram the modules.

Jani4ik, 2021-01-23:
I did this, but the problem remained.
Tell me, what is this bin file responsible for diybms_controller_filesystemimage_espressif8266_esp8266_d1mini.bin ?

Jani4ik, 2021-01-23:
I recorded a new video - https://youtu.be/MAOMkW4MDHc
Can't change Bypass over temperature.
Thanks.

Daemon256, 2021-01-24:
Latest release

Tag-2021-01-23-11-33
0df777d
did not solve the problem

stuartpittaway, 2021-01-25:
Okay, got to the bottom of this one.

The controller expects each module to match the "global setting" temperature, if any module doesn't match then the warning is thrown.

To fix, simply set the temperature in the "global setting" to the temperature you want and "save settings".  This sets all cell modules to the same temperature and updates the controller to remember that default temperature.

[image]

Daemon256, 2021-01-27:
Latest release

Tag-2021-01-25-14-42
7abd4d2
Problem solved. But only if you set the temperature to 45 degrees on the modules of the attiny841_V420 version.

Jani4ik, 2021-02-02:
I managed to solve the problem by switching to firmware V420SWAPR19R20

stuartpittaway, 2021-02-03:
V420 boards have a maximum temperature of 45 degrees due to design fault as explained on other post - only use the V420SWAPR19R20 firmware if you have unsoldered and swapped R19 and R20