stuartpittaway/diyBMSv4Code #91: Smoke came out of the modules

donnib, 2021-01-29:
So I started to connect the first modules and suddenly smoke starting coming out of the second module. What may I have done wrong ? Polarity seems correct same does voltage. A bad board ? [image][image]

donnib, 2021-01-29:
Here are two close-ups of the two modules [image]

[image]

Chass90, 2021-01-29:
Looks like you have connected TX to TX and RX to RX!! I'm surprised you didnt have more than just smoke as this would cause a short between cell 1 and cell 2. Please connect the TX of one module to RX of next module, and so on.

donnib, 2021-01-29:
[image]
Hmm how can you tell, i didn't see that mistake but i don't know for sure since i removed and placed them again. What's the risk now ?

Chass90, 2021-01-29:
Your first picture, you have the daisy chain wires going from the bottom right connector of module 1 to the bottom right connector of module 2, then you have the bottom left connector of module 2 going to the bottom left connector of module 3

Chass90, 2021-01-29:
It should be daisy chained right connector to left of next module and so on.

Chass90, 2021-01-29:
As for risk, if its working, you are very lucky, but please be careful in future. TX to RX only

donnib, 2021-01-29:
@Chass90 yes i can see now what you mean, the second module is also wrong, it seems i made a mistake there and as i started putting power on each of them suddenly smoke came out. I was very carreful on the polarity from the battery hence @stuartpittaway warning on the back but i was not aware that swapping TX/RX could brake things badly i mean i am used to RS232 where nothing happens but then it just doesn't work

stuartpittaway, 2021-01-30:
If you did manage to short out the connections and the board is still working, well done, however I would also check that the JST connections are still okay.  These seem to melt and get loose when heated, and you don't want the cables falling out.

Bizarrely, somebody asked this question only yesterday!
https://community.openenergymonitor.org/t/does-the-polarity-matter-for-the-module-serial-connections/16692

closed 2021-01-30 by stuartpittaway

donnib, 2021-01-30:
@stuartpittaway yes indeed bizarre but even more bizarre is that I read that post since I tend to read what happens but obviously I didn't pay to much attention. I was lucky. I did however disconnected the power as soon as I saw smoke.

stuartpittaway, 2021-01-30:
I'll update the silk screen to add that warning.  Perhaps the next modules can have a fuse as well :-)

donnib, 2021-01-30:
That's a good idea !