stuartpittaway/diyBMSv4Code #93: Are these values within normal range ?

donnib, 2021-02-01:
[image]

@stuartpittaway In particular after one day of operation are these as expected :

- OSS
- Round trip time for 16S
- Different package received for some modules.

stuartpittaway, 2021-02-01:
O.O.S means out of sequence, the controller sent a message but never got a reply. This often happens just after powering on the controller. At a value of 2, nothing to worry about.

It takes about 200ms per module for the communications, so 3 seconds is about right. I've got some code changes in next release to speed that up.

If you click on the reset counter button, all the values will be zeroed, leave it running overnight and see what the numbers are, so far they look fine

donnib, 2021-02-01:
Great thx. If you have time at some point maybe you can have a look at https://community.openenergymonitor.org/t/diybms-v4/11292?page=127

closed 2021-02-01 by donnib

stuartpittaway, 2021-02-01:
Which issue are you referring to? The forum link is back in October!