stuartpittaway/diyBMSv4Code #94: diyBMS operation

donnib, 2021-02-02:
Hi @stuartpittaway ,
Am i correct in this assumption : diyBMS doesn't state when to charge or not (that's up to the charger) but merely uses rules when cells or pack is under/overvoltage to signal to charger or something else that it should not discharge or charge anymore of potential danger ?

/donnib

stuartpittaway, 2021-02-03:
Thats correct, in the future using CANBUS or RS485, it may be possible for the controller to talk to the charger or inverter to start/stop them, but as a BMS its job is to act in an emergency situation and kill charging/discharging - normally via some sort of high power relay or contactor/circuit breaker.

closed 2021-02-03 by stuartpittaway

donnib, 2021-02-03:
Great, thx for clarifying.