stuartpittaway/diyBMSv4Code #95: Incorrect temperature readings V420SWAPR19R20

Jani4ik, 2021-02-02:
I downloaded the firmware V420SWAPR19R20 and get an inflated temperature in the web interface.
Screen - http://joxi.ru/vAWppbwCOoq0yr
The actual temperature of the modules does not exceed 45 degrees Celsius in bypass mode.

stuartpittaway, 2021-02-03:
Hello, are you using V4.20 PCB boards and have you physically unsoldered R19 and R20 and swapped them over?

If not, that firmware is not for you.

On V4.20 boards, the maximum temperature is 45oC, because the thermistor is too far away from the resistors generating the heat.  This was a design/build error at the time, and was fixed by V4.21 boards.

Jani4ik, 2021-02-03:
> Hello, are you using V4.20 PCB boards and have you physically unsoldered R19 and R20 and swapped them over?
> 
Hello.
Yes, I swapped R19 and R20.
Thanks.

stuartpittaway, 2021-02-04:
Hi Jani4ik, so are you measuring the boards using an external thermometer ?

You have bypass set to 90 degrees, and the boards are keeping that temperature (it thinks).

Jani4ik, 2021-02-05:
Hi stuartpittaway, in idle mode the module shows a temperature of 40 degrees Celsius, which is ~20 degrees actually.
dashboard - http://joxi.ru/xAeOObatXn84Vm
module - http://joxi.ru/Q2KBBlMivo6kPm
In bypass mode, the module shows a temperature of 90 degrees, and the actual temperature of the module is ~45 degrees Celsius.
dashboard - http://joxi.ru/82QNN0zC9zaJ7A
module - http://joxi.ru/52aZZbdUkaQw7A
I checked the board with a pyrometer.
Thanks.

stuartpittaway, 2021-02-05:
Hello Jani4ik,

The code which handles this is in the file packet_processor.cpp in the ATTINYCellModule project.

Was the thermistor you installed this one?

Murata Electronics NCP18WB473J03RB
47K ±5% 4050K ±2% 100mW 0603 NTC Thermistor

```
#if (defined(DIYBMSMODULEVERSION) && (DIYBMSMODULEVERSION == 420 && defined(SWAPR19R20)))
    //R19 and R20 swapped on V4.2 board, invert the thermistor reading
    raw_adc_onboard_temperature = 1225 - value;
#elif (defined(DIYBMSMODULEVERSION) && (DIYBMSMODULEVERSION == 430 && defined(SWAPR19R20)))
    //R19 and R20 swapped on V4.3 board (never publically released), invert the thermistor reading
    raw_adc_onboard_temperature = 1000 - value;
#else
    raw_adc_onboard_temperature = value;
#endif

```

This code was put in place in [April 2020](https://github.com/stuartpittaway/diyBMSv4Code/commit/8e214bd7e652efbcb70ed24f455a0952ded1b232#diff-743e206e4c90679ec5d01174e2f33390ea880ebd2f8f2d345b7988a01ea64738)

The "1225" number was increased from 1000 after a user request, perhaps try changing this back to 1000 and upload the code to the module again?

Jani4ik, 2021-02-05:
Hello, stuartpittaway
> Was the thermistor you installed this one?
> 
> Murata Electronics NCP18WB473J03RB
> 47K ±5% 4050K ±2% 100mW 0603 NTC Thermistor
> 
Installed by me Thermistor - http://joxi.ru/l2ZQQbDt7GjEym
> ```
> #if (defined(DIYBMSMODULEVERSION) && (DIYBMSMODULEVERSION == 420 && defined(SWAPR19R20)))
>     //R19 and R20 swapped on V4.2 board, invert the thermistor reading
>     raw_adc_onboard_temperature = [...]
can I ask you to create a bin file for me? I am experiencing a problem with PlatformIO.
Thanks.

stuartpittaway, 2021-02-05:
Here you go, this one has the 1225 changed to 1000.

Let me know how you get on.  Given the mistake on the PCB I wouldn't expect this to be 100% accurate on the temperature but it should be close, with in a few degrees C.

[module_firmware_attiny841_V420_SWAPR19R20_420_SWAPR19R20.zip](https://github.com/stuartpittaway/diyBMSv4Code/files/5932046/module_firmware_attiny841_V420_SWAPR19R20_420_SWAPR19R20.zip)

Jani4ik, 2021-02-07:
Hi, Stuartpittaway.
Is working, the temperature is close to the actual one.
Thanks.

closed 2021-02-08 by stuartpittaway