stuartpittaway/diyBMSv4Code #96: Werid problem on the same spot even after changed the cell

donnib, 2021-02-04:
Hi,
I thought i had a bad cell since the voltage went to high on it so i thought the cell is bad so i replaced it and then i had same issue again on the same cell. Now i changed the module. Before i changes i looked at the voltage reading from diybms and the voltmeter and they seemed to be the same so i am puzzled what the problem could be. Any ideas ? 

/donnib

stuartpittaway, 2021-02-04:
Check the power connections to the module/JST cable and also the cables to the left and right of this module in the chain.

donnib, 2021-02-04:
@stuartpittaway you mean all the cables to the left and right of the cell or do you mean just the power cables ?

What happened is that this module got to 3.5v (bypass voltage) then the bms tried to keep it down but could not so it got to 3.7v and i stoped the charger entirely since the rule in bms was 3.6v but i only configured the charger to go into float and in float some charging still came in that's why it go to 3.7v. Anyway i disconnected everything then started to burn energy of this cell to get it down to a reasonable level. 

All other cells seems to have balanced fine, bear in mind this is first charge of the pack. Here you can see it's almost there. When done i will connect the charger again but i think before that i will check the cables and maybe measure the voltage to see if there is any calibration needed. If you have other suggestions let me hear.

[image]

fisht99, 2021-02-04:
The attached picture shows cell 1 in bypass at 3.35v with a maximum of 3.38v
Is this correct?

donnib, 2021-02-04:
Yes this is correct. I was trying to balance manually to get the cell down to match the others before i start the charger again. I can't get the voltage up on the pack to 56,8V for Absorption before one cell triggers at 3.6V to stop charging. I don't know what i can do to get it in balance, maybe i need to charge lower ? I am charging at 35A now which is 0,22C.

donnib, 2021-02-04:
[image]

donnib, 2021-02-04:
I have moved around the module and exchanged them around but as soon as i charge the range get's higher and it's always on cell 1 that stops the charger

stuartpittaway, 2021-02-04:
Drop the current on the charger and trickle charge them until full

donnib, 2021-02-04:
> Drop the current on the charger and trickle charge them until full

I was just thinking of that. I will try that right now, thx.

donnib, 2021-02-04:
@stuartpittaway But I think your first comment about the cable is most likely right. I think the power cable for cell two is not crimped well or the contact is bad because I can see it's not capable of bypassing quick enough compared to others. It's always this cell that triggers stop of the charger however trickle charger with lower current seems like the way to go after I solve the cable.

donnib, 2021-02-04:
[image]

donnib, 2021-02-05:
I think trickle charge will fix the problem hopefully.

closed 2021-02-05 by donnib