stuartpittaway/diyBMSv4Code #97: Often reboot of controller when trying to access the web interface

donnib, 2021-02-05:
I have seen this 5-6 times that the controller restarts when user want to access the web interface. I don't know what causes this.

stuartpittaway, 2021-02-05:
This is likely to be related to #98

Often the controller restarts are a problem of the power supply or 3.3v regulator on the ESP8266.

Cheap chinese clones are known to have poor quality or incorrect specification regulators, when the ESP8266 tries to communicate over the Wifi a lot more power is needed, the voltage drops and the chip reboots.

It can also be a code issue, but without debug information its impossible to say.

donnib, 2021-02-05:
@stuartpittaway i think it has something to do when it does more on wifi, i am using the webpage extensively now and sometime i switch between phone and two different computers so maybe it get's overloaded.

stuartpittaway, 2021-02-05:
You can't do that, it only works on one web browser at a time - otherwise you will run into issues!

donnib, 2021-02-06:
Got it !

donnib, 2021-02-11:
I see this quite a lot. As far as i can see no other is using the Web interface but i keep changing between different machines or phones for example i would look on one machine, shut it down then try to access from iOS -> bang reboot or the same but from a PC. It looks to like there is a session on the ESP that is left floating somehow and not cleaned up. I think there might be a bug somewhere. I try to behave nicely to it but still. 

Bear in mind that ESP is probably at close in performance since i am running MQTT and influxDB, maybe that's causing this.

MatthiasN73, 2021-04-11:
Hello, 
first sorry for my english and a big thank to Stuart for this project.
I can confirm this issue exactly. 
My controller used the firmware from 30th july 2020 for an half year without any problem.
A couple of weeks ago i updated to the latest version from git and the problems started.
I think its not a hardware problem, its the same.