stuartpittaway/diyBMSv4Code #98: Can't save settings

donnib, 2021-02-05:
I have seen this issue few times.

[image]

stuartpittaway, 2021-02-05:
This is by design.  It prevents cross site scripting attacks, this shouldn't be a problem unless the controller has been rebooted and you didn't refresh the webpage.

donnib, 2021-02-05:
@stuartpittaway the last thing might have happened yes. Maybe the wording could be, Failed to save settings, please make a refresh or something like it ?

jbuszkie, 2021-03-31:
I just hit this too...  Refreshing the browser fixed it.  I agree with @donnib.  Adding something to the warning might help.

stuartpittaway, 2021-03-31:
This occurs when you reboot the controller, but don't refresh the web page

jbuszkie, 2021-03-31:
Yeah that's what happened.  But maybe you can change the "Failed to save settings" to include something like "Failed to save settings. Try refreshing the browser"  Or something like that...

Thanks!