stuartpittaway/diyBMSv4Code #99: More rules for different voltages

donnib, 2021-02-06:
Hi @stuartpittaway 
During my setup i was thinking it could be extremely useful to have one more rule for overvoltage / undervoltage. In my case i use two relays towards the Victron inverter, one to tell it the battery is full = disable the charger e.g on overvoltage of the cell or the pack and another one for undervoltage of the cell or the pack to swich the inverter to charger only BUT in case something goes wrong, the inverter doesn't get the signal or something unforeseen happens then i'd like to have a extra rule that goes a bit lower or higher on the voltage which can trip the circuit breaker for the whole battery, this way i have another measure for security.

/donnib

stuartpittaway, 2021-02-08:
Hi Donnib, doesn't the Victron inverter stop charging automatically at its set voltage?

Why does the BMS need to tell the charger to stop? (unless in a critical situation)

donnib, 2021-02-08:
It does BUT when a cell reaches overvoltage the inverter is told to stop charging obviously the inverter doesn't know the state of each cell so from it's perspective everything seems fine when one cell in-fact is not. Usually one would also have some smaller DC consumers that could drain the battery dangerly low and in order to prevent the continuous drain one could trip the circuit breaker e.g cutting all consumers out.

For me during both the charge and discharge i had the BMS turn the charger off or on since the batteries are not in balance yet. In a normal case the BMS should be the last option, the charger should stop at the given voltage but i doubt that happens in real life for use with self build powerwalls. I have a range of 95mV now on my 16S of 160Ah in rest and with load it can go between 130-200mv which at some point one cells triggers the overvoltage/undervoltage rules. 

Also i noticed the rules take some time so in my case i have set the overvoltage for my LiFePo4 to 3.6V but some cells do reach 3.68V because when the cell go down and the charger is started again with full throttle then it takes a little time before the BMS triggers the rule again and in those few sec the battery voltage can raise above the threshold. I guess this is because of the fact that roundtrip is quite high, i know this will improve in the future.

donnib, 2021-02-09:
I guess the simplest answer to your question is. More rules to have extra safety net, if the charger doesn't stop or behave then have an extra option to pull the plug.