stuartpittaway/diyBMSv4Code #101: Controlled stopped working during night

donnib, 2021-02-11:
Hi @stuartpittaway,
Have you or some other users experienced the controller can stop working and it doesn't come up? I noticed this morning that i could not reach my controller however in the troubleshooting i can see it stopped sending influx data extremely close to the time i can see the WiFi AP had a reboot or some kind since the duration since last restart (AP) fits with the controller "dying". 

What does the controller do if it can't find the AP ? Does it try again ? If so for how long ? 

/donnib

stuartpittaway, 2021-02-11:
Hello @donnib, just to clarify:

1. Your wifi router rebooted overnight
2. Controller didn't automatically reconnect to the wifi when it came back up

Do you know if the controller was still operating correctly (green lights on the modules for example) ?

donnib, 2021-02-11:
1. Yes the AP did reboot and came up few minutes after (seeing from the logs).
2. Yes. I checked the WiFi list clients list and it wasn't in the list and i could not reach it. 
3. Yes the green lights were flashing indicating that it had a connection to the controller, i also presumed that.

stuartpittaway, 2021-02-11:
Ok thanks.  There was a problem a while back where the controller would reboot if the WiFi was lost, it looks like this isn't a problem which is good.

The controller should retry to connect to WiFi indefinitely, however its possible this code is wrong or the ESP8266 didn't realise it was disconnected.

I'll review the code and see what I can find.

stuartpittaway, 2021-02-11:
I think the existing code would attempt to reconnect to the Wifi only once before ignoring it completely.  I've changed this behaviour to attempt to connect every 30 seconds.

This is in check in  475d46adbca58d94fb78369c6bf12148c1cc31c3

closed 2021-02-11 by stuartpittaway

donnib, 2021-02-11:
@stuartpittaway Great work. I think i saw some code regarding OTA updates, is that possible or do i need to do an update over USB ?

stuartpittaway, 2021-02-11:
The code is in the master branch, along with a few other fixes, OTA is enabled, but I find it quite hit and miss!

The password for the OTA is here...

https://github.com/stuartpittaway/diyBMSv4Code/blob/77957bb6525792051bea6cafd37511b7f4c84217/ESPController/src/main.cpp#L767