stuartpittaway/diyBMSv4Code #106: Status relays

pablo170, 2021-03-24:
Stuart congratulations for your amazing project, is it possible to add a delay to the relays or something similar while diybms starts? 
I think who chooses this bms with victron inverter set the two signals bms in configuration, but what happens is that if we restart  the bms, while diybms starts searching the modules, the relays communicates with victron like the battery is empty and the inverter starts to charge in sustain mode. Could be there any enhancement for fix it? Thanks in advance.

stuartpittaway, 2021-03-24:
When the controller is reset, it cannot control the relays so they switch off.

No code is running at that point, so I cannot affect the relays at that point.

pablo170, 2021-03-24:
Thank you Stuart for you fast answer, I try a lot of examples to understand why the relays were activated or deactivated. I used NC and NO contacts and default status on/off relays, and I think if the relay is in idle mode, and when the diybms is restarted the relay is activated while the message waiting for reply modules is shown, isn't it possible that the controller interpret the message like if the voltage value is under the set point and it is why it sends the signal to the relay? Thanks in advance.