stuartpittaway/diyBMSv4Code #111: Is there a temperature calibration?

CrAzYDr1veR, 2021-05-05:
Hi, normally my modules are like for example at 17 and others 18 degrees.
I got some new external temperature sensord but they range from 14 to 17 degrees (different values in multimeter).
Is there a calibration like the voltage so all sensors can read the sabe value and start from the same point?
[image]

stuartpittaway, 2021-05-05:
What sensors are they? If they are a different resistance or K value then the readings will be incorrect.

CrAzYDr1veR, 2021-05-05:
same k value 47k B parameter is 4050K as you can see some have 17º just like the ones on the v4 board, others are of even when mesuring with the multimeter.
NXFT15WB473FA2B150 47k 4050K
https://community.openenergymonitor.org/uploads/short-url/owZzXetv8cStkZp09J5U2lHGOL0.pdf

stuartpittaway, 2021-05-06:
Ok, are these v4.0 or v4.2 boards? The V4.4 swapped to using different thermistors.

There isn't currently any way to calibrate these, as that is that the 4050K calibration value does!

Is the resistance ok when checked with multimeter? Should be 47K at 25degreeC

CrAzYDr1veR, 2021-05-06:
4.0 boards. At 18 degree the resistance value os some is different than others so i imagine that is the problem.
So i should try to change this calibration value to see if i ca get closer to the real temperature.

stuartpittaway, 2021-05-06:
You can try, there are values in platform.io, and you will need to recompile the code manually.

CrAzYDr1veR, 2021-05-06:
ok thanks

closed 2021-05-06 by CrAzYDr1veR