stuartpittaway/diyBMSv4Code #112: No connection after fw update

jency001, 2021-06-06:
Hi Stuart,

I wanted to update the esp8266 module because I wanted it to connect to another network. So I reinstalled the latest fw but the esp does not find any modules. Is is because the fw on the board is older than that of the esp?

Kind regards

stuartpittaway, 2021-06-06:
It should work, is there a red message bar saying Found X modules?

jency001, 2021-06-06:
[image]
i ngot tx to rx connected and vise versa. it did work befor and now i am trying to get one working but no connection also no green leds blinking

stuartpittaway, 2021-06-06:
Ok, connect tx to Rx on the controller only, no modules. Do you get an error count number  increase?

jency001, 2021-06-06:
yes i do

closed 2021-06-06 by jency001

jency001, 2021-06-06:
also intressting when i plug in rx on the module from tx from the controller the blue led stops blinking no leds are blinking now. en wehn i remove the cable te led blenks every 10 sec

stuartpittaway, 2021-06-06:
How old is the firmware on the modules?

jency001, 2021-06-06:
more than 1 year

stuartpittaway, 2021-06-06:
Ok, it's likely very old compared to the controller. So you would need to flash the modules as well