stuartpittaway/diyBMSv4Code #113: Only 2 projects tasks

Bogdancev, 2021-06-08:
Hi Stuart
I followed your video and downloaded/opened the project in PlatformIO, but contrary to what is in the video I only see 2 tasks, there is no controller task, nor other module versions tasks.
Is this correct or my setup is wrong or the project has chanced since the video was produced?

[image]
Regards, Sergii

stuartpittaway, 2021-06-08:
The video is a little out of date now as platformio keeps changing.

You can use the pre-generated files from GitHub, take a look at the instructions there.

Alternatively, edit the platformio.ini file and remove any of the module versions you don't want.

Bogdancev, 2021-06-09:
Above happened when I opened a workspace as in the video, and after looking closer I saw that "attiny841_V400" was a task from "ATTINYPIDTuning" project. That project is the top one in the list on workspace projects.

Then I used "File -> Open Folder" instead of "File -> Open Workspace" and opened "ATTINYCellModule" folder and after that all tasks of different module verstions became visible.
[image]

So I'm not sure maybe there is a way to specify which project of workspace projects is active at the moment or maybe there is another way to control which project's tasks are shown.

I will use it as "Open Folder", it builds fine.

closed 2021-06-09 by Bogdancev